supermerill/SuperSlicer #1538: Higher speeds being forced for first object layer when using Raft with Solid 1st Layer set

gpread, 2021-09-09:
2.3.56.8 & 2.3.57.0 (didn't check but its surely probable therefore that 2.3.56.9 is afflicted also)

Creality CR-10, Marlin 2.0.8.2
Windows 10 - 20H2 (19402.1165)

First Object Layer speed behaves differently as soon as  the options for a raft with a solid 1st layer is applied

In the specific case I observed this it seems to be applying/forcing the thin walls speed to this layer when it was not (or at least not in the same way) until both solid first layer with raft is applied.  Raft alone does not trigger this behaviour.

I suspect that Raft with solid first layer might also trigger changes to speeds in  g-code for object layers for other scenarios (e.g. where thin walls don't apply) and therefore that thin wall speed may not be a means to manage this issue. 

1. load model/open project (provided below)
2. Make sure no raft or raft without solid first layer is selected
3. Slice the model
3. Look at the layer times and/or speed views, noting how they are for the object layers
4. Enable Raft with solid 1st layer
5. Re-slice the model
6. Review the layer times and/or speed views for the object layers (it should be seen that the 1st object layer speeds have changed)

[image]
Speeds with raft but no solid 1st layer

[image]
Layer times with raft but no solid 1st layer
 
[image]
Speeds with raft and solid 1st layer

[image]
Layer times with raft and solid 1st layer

In my specific case the speeds and layer times are (and should be) being influenced by the short layer time goals in filament settings.  Therefore another aspect of what is seen to occur here is that those goals are not being equally respected when raft and solid 1st layer are applied.

gpread, 2021-09-09:
[1st Object Layer speed increase triggered by raft&solid 1st layer_issue#1538.zip](https://github.com/supermerill/SuperSlicer/files/7134625/1st.Object.Layer.speed.increase.triggered.by.raft.solid.1st.layer_issue.1538.zip)

supermerill, 2021-09-09:
in your config:
slowdown_below_layer_time = 60

so it tries to increase the layer time up to 1 min.
That's why you have a 1 min layer time.

gpread, 2021-09-09:
I don't complain about the layer times (but them reducing is also evidence that speeds were increased).

Yes it is slowing the speeds down in order to increase the layer times (even if the low speed limits don't allow it to get them above 1 minute.

However, it is increasing the speed and reducing the layer times as soon as a raft with a solid first layer is requested.

To me that clearly does not make sense/is undesirable (object layer speed behaviours should not change just because some support was added in the layers below) and also the increase in speeds seen (1st object layer goes from 10mm/s to 30mm/s)  is clearly also moving against the short layer time goals existing in the filament settings (so they are not being equally respected).

supermerill, 2021-09-26:
I may be able to change that to use the layer above instead of the one below.
But i need to wait my gcode writer rewrite to do that.

gpread, 2021-09-26:
FYI, I noticed something whilst looking at PS 2.4.0 alpha. In Print Settings->Speed->Modifiers, they have included an option to set the "Speed of the object first layer over raft".  This becomes enabled once raft is selected and for me automatically defaulted to 30mm/s.  So perhaps the behaviour in SuSi is due to something inherited from partial implementation within PS ?

Seems to me that they should make it default to whatever has been set in first layer speed (absolute), currently with default profiles (no set up/configuration wizard, or inherited profiles) "First layer speed" was 20mm/s and  "Speed of the object first layer over raft" was 30mm/s.

supermerill, 2021-09-26:
>So perhaps the behaviour in SuSi is due to something inherited from partial implementation within PS ?

It's something new, as they overhaul the supports/raft code.
This should be merged back into susi.