supermerill/SuperSlicer #1539: Overlapping External Perimeters can cause perimeter gaps (extreme under extrusion) as opposed to what's warned (potential over extrusion)

gpread, 2021-09-09:
2.3.56.8 & 2.3.57.0 (didn't check but its surely probable therefore that 2.3.56.9 is afflicted also)

1. Open project (file provided below)
2. Check that Print Settings->Perimeters & Shells->Quality->Overlapping external perimeters is disabled (not the default which is enabled)
3. Slice the model and view the feature type preview, note that the model elements are complete (no gaps in the walls)
4. Set Print Settings->Perimeters & Shells->Quality->Overlapping external perimeters to enabled
5. Re-slice the model and view feature type preview, note that some elements of the model now have gaps in the walls)

Obviously this only occurs where there are (or would be some) thin wall scenarios (those config options are specific to such).

However, besides that its currently unexpected/unpredicted and undesirable to end up with gaps (or to loose the benefits of more than a single perimeter, by disabling overlapping perimeters. 

Given the nature of the model (and that I checked X-Y compensation didn't influence the outcome) I don't understand why the gap problem does not occur on the 1st layer (only the layers above it) ?

Since the 1st layer width by config is slightly larger than the perimeter width, if anything, I would expect it to gap on the 1st layer and not the others or artefacts to be worse on the 1st layer (not better as they observed to be).

[image]
Feature Type view with Overlapping External Perimeters disabled (non default)

[image]
Width view with Overlapping External Perimeters disabled (non default)

[image]
Feature Type view with Overlapping External Perimeters enabled (default) - showing gaps in the walls (but not on the 1st layer)

[image]
Width view with Overlapping External Perimeters enabled (non default) - showing that in the region of the gaps extrusion is significantly reduced (but not nearly as much on the 1st layer)

gpread, 2021-09-09:
[Overlapping External Perimeters can cause perimeter gaps_issue#1539.zip](https://github.com/supermerill/SuperSlicer/files/7135253/Overlapping.External.Perimeters.can.cause.perimeter.gaps_issue.1539.zip)

supermerill, 2021-09-28:
failed thin wall issue.