supermerill/SuperSlicer #1541: Smarter auto-arranging

kyleisah, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
Not related to any problem, just a feature request!

**Describe the solution you'd like**
I would like models on the plater, when auto-arranged, to "staircase". 

**Describe how it would work**
Ideally the slicer would analyze what models are the tallest and keep them close to objects that are similar in height, as with smaller objects. This would save a lot of time in unnecessary travel by not traveling over finished models on the build surface. The slicer can already detect the size of objects, so some of the work is already done. Maybe like an "Arrange plate by" dropdown with "height" or "width" or "speed" or "cooling speed" etc. Would be great to have that kind of control when printing a lot of different parts, or even keeping parts that need more cooling away from parts that need less, to prevent warping parts.

**Describe alternatives you've considered**
Manually arranging the shortest part in the center, and working my way back with taller parts so there's no unnecessary travel moves.

supermerill, 2021-09-26:
I won't work on it.
it's something where a human is a lot more efficient and can do the work manually.

ldiegos, 2021-09-27:
Hi,
@supermerill going on the re-arrange objects i rather prefer that susl arrange the objects in the bed following the parameters of distance between objects and clearance radius, if is checked the "Complete individual objects"

[image]

Because, right now, maybe i don't know how to do it, but when i drop a object in bed, it automatically goes to center and all objects collide on g-code preview and i have to start moving objects until the "warning message" disappear.

Maybe at this point, the computer eye, do better work than the human eye, using more efficiently the bed space.

Just a suggestion on re-arrange stuff :)
Luis.

supermerill, 2021-09-27:
@ldiegos 
What you describe isn't the current arrange function (fourth button of the 3D view, next to the trash can) ?

ldiegos, 2021-09-27:
> @ldiegos
> What you describe isn't the current arrange function (fourth button of the 3D view, next to the trash can) ?

yuup... yes... 🥇  

 i never use those buttons... sorry(what a fail from my side). Is my first time that i see them, btw lol