supermerill/SuperSlicer #1543: M204 being emitted unless acceleration control is enabled then disabled

RealDeuce, 2021-09-10:
**Describe the bug**

An M204 is being emitted to GCode if you haven't enabled then disabled Acceleration Control in v2.3.57.0 on FreeBSD.

I set acceleration in my custom GCode and if I don't enable/disable acceleration control before slicing, the acceleration is changed after my custom gcode to a value that I'm not even sure where it's coming from.

**To Reproduce**
Steps to reproduce the behavior:
1. Open SuperSlicer
2. Set Printer Settings -> Machine limits -> How to apply limits to "Use also for time estimate"
3. Save Printer Settings
3. Set Print Settings -> Speed -> Acceleration control (advanced) -> Default to 0 (disables Travel setting)
4. Save Print Settings
5. Exit Super Slicer
6. Open Super Slicer
7. Add object to print bed
8. Slice and export
9. G-code contains an M204 just before "move to first Skirt point (acceleration)"
10. Set Print Settings -> Speed -> Acceleration control (advanced) -> Default to 1 (enables Travel setting)
11. Set Print Settings -> Speed -> Acceleration control (advanced) -> Default to 0 (disables Travel setting)
12. Slice and export to a different file
13. G-code *does not* contain an M204

Whatever the difference is, it does *not* show up in the .3mf file either... the contents of a before/after 3mf file are identical aside from the "generated by SuperSlicer" timestamp.

**>> Project File <<**

Example included.  If you open this 3mf and slice, you get an M204.  If you enable/disable acceleration control then slice, there is no M204 (the first gets an "M107 ; disable fan" after the start_gcode as well for some reason)

I've included a 3mf saved before enable/disable (InvalidM204.3mf) and after (ValidM204.3mf).  After saving each 3mf, I generated G-code which is also included (InvalidM204.gcode has the M204 and M107, ValidM204.gcode has neither)

[M204Issue.zip](https://github.com/supermerill/SuperSlicer/files/7143068/M204Issue.zip)

**Expected behavior**
Change a setting, then changing it back should not change the generated G-code

**Desktop (please complete the following information):**
 - OS: FreeBSD 12
 - Version 2.3.57.0

RealDeuce, 2021-09-10:
Actually, it looks like *any* change which allows you to re-slice eliminates the M204.

The *first* time after starting the program that you slice an object, the M204 is emitted and the fan is adjusted.  Every time after that, the M204 and M107 are *not* included.

This looks like a global variable tracking acceleration and fan speeds that is not initialized properly.

RealDeuce, 2021-09-10:
Changing *some* settings will not trigger the M204 going away... setting perimeters then changing back for example does not seem to make a difference... after doing that, the M204 remains.

RomRider, 2021-09-13:
Same, on windows.
M204 S500 is emitted before first skirt line.

supermerill, 2021-09-27:
maybe a duplicate of https://github.com/supermerill/SuperSlicer/issues/1537

can you try with the latest nightly?

RealDeuce, 2021-09-30:
With the latest Nigthly, the M204 that irritates me isn't emitted anymore, bit there's a harmless M104 and M107 in the initial slice that aren't there after the steps above.

`> diff -u ManyParts-before.gcode ManyParts-after.gcode
--- ManyParts-before.gcode	2021-09-29 23:41:49.098898000 -0400
+++ ManyParts-after.gcode	2021-09-29 23:42:37.429664000 -0400
@@ -1,4 +1,4 @@
-; generated by SuperSlicer 2.3.57 on 2021-09-30 at 03:41:01 UTC
+; generated by SuperSlicer 2.3.57 on 2021-09-30 at 03:42:09 UTC
 
 
 ;
@@ -433,10 +433,8 @@
 SET_PRESSURE_ADVANCE ADVANCE=0.07215
 ;M900 K0.08
 ; custom gcode end: start_filament_gcode
-M104 S230 ; set temperature
 M109 S230 ; set temperature and wait for it to be reached
 M190 S70 ; set bed temperature and wait for it to be reached
-M107 ; disable fan
 ;LAYER_CHANGE
 ;Z:0.2
 ;HEIGHT:0.2
`

supermerill, 2021-09-30:
>  there's a harmless M104 and M107 in the initial slice

If you want full control of the start gcode, you can check the `only custom start code` option next to it.
If you do that, don't forget to also put 
```
M107 ; disable fan
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
```
into it.

RealDeuce, 2021-10-02:
Yeah, everything I care about is fixed, but the differing behaviour between the first slice and another slice after changing the settings, then changing the back is weird.

commit b66f41f4 (2019-01-25): Fixed #1543
closed 2021-10-02 by RealDeuce