supermerill/SuperSlicer #1544: How to find the line in the gcode which corresponds to the tool position (percision issue)

laptop-laptop, 2021-09-10:
**Is your feature request related to a problem? Please describe.**

Hi, I used a mouse to move the horizontal slider to the left and right to find the range of positions that I want to modify the gcode. For example, the slider shows: 412613 and 415170 as the starting and ending points where I would like to issue a stop to allow filament change and perhaps also temperature change. The tool positions are:

412613: X140.02, Y:129.05, Z:5.05
415170: X126.48, Y:114.95, Z:5.08

I tried to use the search function in Atom the text editor. However, the positions in the gcode have 3 decimal places while those shown in the tool positions in the slicer have 2 decimal places. As  a result, I cannot find the matching lines in the gcode.

How do I find the corresponding lines in the gcode? 

Is there any way that I can make the gcode to output numbers in 3 decimal places? Not sure if that would mess up with the printer. Alternatively, can we make the Tool position to display numbers in 3 decimal places so that I can search the corresponding lines in the gcode using the search function of the text editor?

I went to Configuration->Preferences but I don't seem to find such options in precision.

ldiegos, 2021-09-11:
Hi,
Superslicer allow you to change both precision, movement and extrusion:
[image]

At least you can change to 2 decimas to match what you want, but i don't know how will affect the losing of the third decimal to the overal result.

laptop-laptop, 2021-09-11:
Hi, thank you. It is nice to know such feature exits. I am concerned about how losing the third decimal could affect the overall result. Anybody knows? I do know that after upgrading to the latest version, the XY compensation for vertical holes setting I made for 2.2.53 do not work perfectly. I need to change the value at the third decimal as far as I recall. So I worry about potential side effects.

gpread, 2021-09-12:
You might want to look at #1555 as I think this is perhaps what you might be looking for ultimately ?