supermerill/SuperSlicer #1545: Only One Perimeter on Top Surfaces is applied to 1st layer - causing narrow infill for manual/custom supports

gpread, 2021-09-11:
2.3.57.0 

Creality CR-10V3 (modified), Marlin 2.0.8.2
Windows 10 - 20H2 (19402.1165)

When "Only one perimeter on Top surfaces" is enabled (Print Settings->Perimeters & Shells->Quality) it is triggered for the first layer. This can result in single a perimeter with narrow infill for any manual support structures which utilise a single layer which is not good for adhesion or printability.

[image]
With "Only one perimeter on Top surfaces" disabled (non-default)

[image]
With "Only one perimeter on Top surfaces" enabled (default)

Surely this should not trigger on the first layer, or perhaps even better not trigger for any surface which is only one layer thick ?

1. Load the project file (attached below)
2. Make sure Print Settings->Perimeters & Shells->Quality->Only one perimeter on Top surfaces is unchecked/disabled
3. Slice the model, then examine the Feature Type view and note that the base of the model has multiple perimeters that make it up
4. Go to Print Settings->Perimeters & Shells->Quality->Only one perimeter on Top surfaces and enable/check it
5. Re-slice the mode, then examine the Feature Type view and note that the base now has only a single perimeter and some narrow infill area.

gpread, 2021-09-11:
[1 Perim on Top Surfaces is applied to 1st layer_SS_Issue#1545.zip](https://github.com/supermerill/SuperSlicer/files/7147328/1.Perim.on.Top.Surfaces.is.applied.to.1st.layer_SS_Issue.1545.zip)

supermerill, 2021-09-15:
> Surely this should not trigger on the first layer, or perhaps even better not trigger for any surface which is only one layer thick ?

I don't know why. If I have to print a writing to glue onto something and it use colorprint to switch color on each layer, it's useful.

For your issue, why not using a brim?

gpread, 2021-09-15:
Its part of a larger print and I'm using internal brim on the main part, which in fact does not work on the support either (it causes ears on the support (see other issues I raised, which ultimately were all discovered from one project but I broke the issues down for easier tracking and resolution into separate issues and project files)

supermerill, 2021-09-16:
"not-X at first layer" should be taken care of by #412 when it's done.