supermerill/SuperSlicer #1550: Additional extraneous perimeter part added above current layer perimeter at the end of current layer

gpread, 2021-09-11:
2.3.57.0

Creality CR-10V3 (modified), Marlin 2.0.8.2
Windows 10 - 20H2 (19402.1165)

[image]
Feature Type view showing an extraneous perimeter part added at the end of current layer (its no longer visible if/when you step up to the next layer).

This happens on multiple/many layers (can seen by using the vertical slider) but not ever layer.

https://user-images.githubusercontent.com/59426970/132940800-b6cf4a8f-5a98-4d9b-8471-2948a226f111.mp4
~40 sec screen capture recording of using the tool position sliders that shows how its moving up a layer within and at the end of the current layer to add the extraneous perimeter part. 

1. Load the project file (attached below)
2. Slice the model, then examine the Feature Type view and use vertical slider to select say layer # 27 (5.4mm height)  and note that there is an extraneous perimeter part sat on top of the current layer
3. User the horizontal slider to move the toolhead back and you can see that its adding this at the end of the current layer after it has completed the perimeter already.
4. If you examine in a similar way other layers it can be seen that the length of the extraneous parts is highly variable, in some cases it approaches 50% on a complete perimeter!

gpread, 2021-09-11:
[Extraneous perim part added above current layer perimeter at the end of it_SS_Issue#1550.zip](https://github.com/supermerill/SuperSlicer/files/7147429/Extraneous.perim.part.added.above.current.layer.perimeter.at.the.end.of.it_SS_Issue.1550.zip)

supermerill, 2022-02-01:
gui issue