supermerill/SuperSlicer #1554: "Advance length" setting for overhang perimeters to adjust speed prior to reaching the overhang

gpread, 2021-09-11:
For overhangs provide the ability to include config for an "advance length", that is - specify a mm distance prior to the overhang that adjusts perimeter speed(s). 

Currently using region modifiers does not represent a workable solution even since they result in completely new perimeter paths being formed creating new seams and points of weakness, rather than just a new segment in the existing perimeter at the new speed. 

Region modifiers is a weaker approach even if they didn't create new perimeter segments as modifiers would need to be defined for every overhang location, which in many cases would also be hard to accurately define.

Such a setting might also be used in combination or alternatively for Fan speed of overhangs rather than the speedup option.  So change in fan speed is triggered at a distance in advance from the overhang rather than a specific time in advance.

supermerill, 2021-09-15:
> For overhangs provide the ability to include config for an "advance length", that is - specify a mm distance prior to the overhang that adjusts perimeter speed(s).

The problem is that the firmware can't slow down quickly enough?
Everytime a speed change is occurring from fast to slow, it needs to occur X seconds before?

gpread, 2021-09-15:
In general, yes I would say that is a fair way to describe it.  Basically with small parts, and/or larger ones with a narrow overhang, I see that the overhang perimeters (and often also internal perimeters - since its small it is effectively solid perimeters in that part) are not well controlled unless I really slow the speeds of perimeters down significantly.

Kinematics/physics mean that deceleration is less well controlled than acceleration, so I believe this would be a useful feature.

My thought was that the perimeter would become treated as an overhang (at least from a speed perspective) before it is technically actually an overhanging perimeter.

So yes either an advance time or distance would be an appropriate way to configure it.  I'm not sure if time is better, for sure its influenced by speed (with regard to where on the perimeter it would take effect) and perhaps that's therefore beneficial. 

I understood your previous point about flow adjustment but at the end of the day the mechanics of the system mean momentum results in nothing occurring instantaneously (which the software assumes with perhaps the exception of corners e.g. Junction Deviation and/or Jerk).

Thus I think a lot of what the slicer and/or the user is doing with the slicer is to predict the future for the firmware/kinematics.  

Currently the analogy I see is that the software/firmware is like the driver of the car where the visibility is near or in fact zero.  As we all know if we drive or fly (I can/have done plenty of both) its important/valuable to have visibility and to learn to anticipate to be able to optimise the performance.