supermerill/SuperSlicer #1555: Easy navigation from views to a specific location in the resultant g-code

gpread, 2021-09-11:
Provide the ability to navigate from the current position of the toolhead within a view (which can/is adjusted using the horizontal and vertical sliders) to go directly to the relevant location in the resultant g-code.

Pure elegance would be to say right-click or double click mouse pointer at the location in the view and open in a text editor the g-code at the appropriate point (i.e. the line within it).

Easier to implement options would be to include a reference on the view of the corresponding line number in the gcode for the current location of the toolhead. If that was available I imagine it would then be pretty easy to enhance it slightly more allowing a double click or similar on that "line number" to invoke a text editor with the g-code open at the appropriate line (notepad++ for example allows command line invocation to pass specific line number as an argument).

I believe this can be a very helpful feature to reduce efforts needed with debugging, research and/or making fine tweaks to production print g-code.

CCS86, 2021-09-11:
I would really like this too. Repetier has this feature.

supermerill, 2021-09-26:
I'll see if I can show the line number somewhere (bottom or popup)

gpread, 2021-09-27:
I can see that since I raised this, they have pretty much added this to PS 2.4.0 Alpha 2.  When you move the horizontal slider for tool position in layer, it shows an overlay of the gcode in the feature type preview !! :-)

So perhaps take a look at that at least before doing anything and/or only finalise once PS 2.4 is merged

nmw01223, 2022-11-28:
Also see #3461.