supermerill/SuperSlicer #1556: Files created using Export G-Code have incorrect time and date 2.3.56.9

laptop-laptop, 2021-09-11:
**Describe the bug**
This is really strange. Files created when I clicked Export G-Code have incorrect time and sometimes even date.
For example, I exported a gcode file on September 11 at 10:00am. In the folder where it was saved, Windows 10 showed  that the file was created on September 11 at 8:50am. Sometimes even the date is wrong. For example, saving the gcode on September 11 at 12:00am and the file system showed that it was created on September 10 at 10:30.

**To Reproduce**
Steps to reproduce the behavior:
See above

**>> Project File <<**
Any file will do

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows] Windows 10
 - Version [e.g. 2.3.55.5], Use `Help->About`_   2.3.56.9

supermerill, 2021-09-26:
what is your Time zone and your Locale?

laptop-laptop, 2021-09-27:
EST in North America. This is very strange but it did happen on the day I reported this issue.

supermerill, 2021-09-28:
The slicer don't do anything with the gcode file, it just asks the os to create it, add some characters inside it and then rename/move it.
I'm wondering what's wrong with macos