supermerill/SuperSlicer #1558: Can we load gcode to SuperSlicer and see how the virtual nozzle moves in the visualization?

laptop-laptop, 2021-09-12:
**Is your feature request related to a problem? Please describe.**

I don't seem to find an option to allow us to load a gcode back to SuperSlicer and then see how the nozzle moves in the visualization. Is there such option? 

**Describe the solution you'd like**
See above

**Describe how it would work**
Just like loading a project file, load the gcode and play back the virtual nozzle movements.

gpread, 2021-09-12:
File->G-code Preview is I think perhaps what you are looking for ?

Full features will only work if the g-code was created by slicer from the same bloodline and possibly the same slicer and version.

laptop-laptop, 2021-09-12:
That is very nice. Thank you.

What you mean by same bloodline?

gpread, 2021-09-12:
> That is very nice. Thank you.
> 
> What you mean by same bloodline?

Something that shares some of the same code (and hence the constructs, labels, comments, etc in the g-code can be common).

In this example/case with SuperSlicer there is at least (could be others too) overlap/heritage from Slic3r and PrusaSlicer

laptop-laptop, 2021-09-16:
In PrusaSlicer 2.4.0-alpha1, when the user slices and starts moving the horizontal slider in preview, it will show a gcode snippet and move the tool accordingly. When will Super Slicer has this feature perhaps also in File->G-code Preview?

supermerill, 2021-09-16:
when prusaslicer code will be more stable, at the end of the beta stage.

laptop-laptop, 2021-09-16:
Thank you. Looking forward to it.

closed 2021-09-16 by laptop-laptop