supermerill/SuperSlicer #1559: Showing the sliced model with differnet colors representing filaments from different extruders

laptop-laptop, 2021-09-12:
**Is your feature request related to a problem? Please describe.**

Got a file created from PS from a colleague. Under PrusaSlicer 2.3.3, I sliced a model and under View: Tool, the visualization showed the main part (made by Extruder1 in orange) and the support (made by Extruder2 in blue). On the top left, under Tool both Extruder 1(orange) and Extruder2 (blue) were listed. It was easy to see which part is made from Extruder1 and which part is made from Extruder2.

Loaded the same file into SuperSlicer 2.3.56. After slicing was completed, it showed the entire thing in orange under View: Tool. Moreover, only Extruder1 is shown under Tool on the top left.

**Describe the solution you'd like**
Just like PrusaSlicer, have different colors to distinguish filaments from different extruders.

**Describe how it would work**
See above

gpread, 2021-09-12:
> **Is your feature request related to a problem? Please describe.**
>
> Got a file created from PS from a colleague. Under PrusaSlicer 2.3.3, I sliced a model and under View: Tool, the visualization [...]
That facility already exists in SuperSlicer.

Not 100% sure but I suspect the reason you didn't see it correctly is because the printer profile you have set up in your SuperSlicer instance has the printer configured as having only a single extruder

laptop-laptop, 2021-09-12:
I clicked Printer Settings. On the left, Extruder 1 and Extruder 2 are listed. Under Printer Settings->General->Capabilities->Extruders, I have 2 and "Single Extruder Multi Material:" with a check mark. Have I missed some settings?

gpread, 2021-09-12:
> I clicked Printer Settings. On the left, Extruder 1 and Extruder 2 are listed. Under Printer Settings->General->Capabilities->Extruders, I have 2 and "Single Extruder Multi Material:" with a check mark. Have I missed some settings?

Seems to me from my own (limited) experience of this area that based on what you just said that it could work.  Maybe its some kind of file/version dependency.  Meaning that SuperSlicer cannot fully migrate/import correctly from the PrusaSlicer originated file.

For sure the feature you asked about is there in SuperSlicer because I have configured my own projects this way and seen it working (different colours representing the filament/extruder changes).

Another thought would be that its to do with the filament profiles, it takes the colour from the setting in the profile for the specific filament being used. So if the same or different filaments were used in the extruders where the colour was the same (guaranteed if same filament used, possible if different filaments used), then on the surface it could look like there is not. Maybe thats were there is something lost in translation.

laptop-laptop, 2021-09-12:
Under Filament Settings->Filament properties->Filament type, there is a description "The filament material type for use in custom G-codes." What does it mean? If I use PLA for the main part and PETG as the support, which one should I put under Filament type?

On the top right, I can see "0.15mm QUALITY (modified)", "Generic PLA (modified)" and "Genetic PETG". However, under Filament Settings, in the pull down manual, I only see "-default'' and "Generic PLA (modified)" listed. 
"Generic PETG" is not listed. Do you think this indicates that there is something wrong with the settings? Shouldn't the pull down manual also have "Generic PETG"? Under PrusaSlicer, I see a list of filaments.

gpread, 2021-09-12:
I suspect that what you are seeing or are not seeing is a result of which view you have selected (in the "view" drop down selector at the bottom right of the g-code preview screen).  Only the vertical bar for layer selection will show filament colours in some views, but there is one specifically for "filament" (others may do it too) that will show the model with the colours applied

laptop-laptop, 2021-09-12:
Under View->Color Print, PS shows two colors and two extruders. Under SS, it shows only "Extruder 1 default color". Under View->Tool, same thing. Under View->Filament, also only Filament1 is shown. There is something wrong I think. Is there an article on how to properly set up multifilament printing using one extruder under SuperSlicer?

gpread, 2021-09-12:
> Under View->Color Print, PS shows two colors and two extruders. Under SS, it shows only "Extruder 1 default color". Under View->Tool, same thing. Under View->Filament, also only Filament1 is shown in SS. There is something wrong I think. Is there an article on how to properly set up multifilament printing using one extruder under SuperSlicer?

Not that I am aware of.  I gained my knowledge from PrusaSlicer, its associated docs, forums, etc and by experimenting with SuperSlicer.  For sure from my experience thus far a weakness around SuperSlicer is user documentation but that's understandable given how its developed

neophyl, 2021-09-13:
you ask for help but not have included the requested copy of your project file that is displaying the issue.  Without it anything is purely guesswork which is a waste of time and energy.  Please supply the needed information to make tracking down any potential problem at least possible.

supermerill, 2021-09-26:
I need the project file.
I suspect a problem from the custom `toolchange G-code`
If there is something inside, it won't emit the `T` command, contrary to prusaslicer, hence why it's printed with only one extruder.
Please clear this field or include the `T[next_extruder]` command