supermerill/SuperSlicer #1560: 2.3.57: Double Travel Between Perimeters

CCS86, 2021-09-13:
I wanted to double check if this was expected behavior. It seems like splitting the move like this could slow it down with jerk/accel limits:

[image]

[Objective Lens Cap.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7150613/Objective.Lens.Cap.3mf.txt)

supermerill, 2021-09-27:
I don't understand.

CCS86, 2021-09-27:
Instead of a single travel move, from the end of one perimeter to the start of the next, it is splitting it up into 2 travel moves.

supermerill, 2021-09-27:
that will add a useless line in the gcode, no?

CCS86, 2021-09-27:
I don't follow. 

I'm saying the the current build, there are 2 travel moves, instead of one. To me it seems like a useless line of code, that could cause motion control to execute it slower than a single move. I wanted to check if you had a specific reason for doing this.

gpread, 2021-09-27:
My intuition makes me wonder if this is related to #1550  ?

supermerill, 2021-09-27:
> I'm saying the the current build, there are 2 travel moves, instead of one. 

weird, I saw only one.

qwewer0, 2021-09-27:
> > I'm saying the the current build, there are 2 travel moves, instead of one.
> 
> weird, I saw only one.

> Objective Lens Cap.3mf.txt

@supermerill Try at Z=7mm, 40298-40300.

supermerill, 2021-09-27:
[image]
something is wrong.
what version?

qwewer0, 2021-09-28:
> it's at 7.2
> something is wrong.
> what version?

I tried it with 2.3.57, but here is a video of it:

https://user-images.githubusercontent.com/57561110/135022170-25bbf535-d906-4a19-a02e-91eb60f85ad4.mp4

closed 2021-11-19 by supermerill