supermerill/SuperSlicer #1563: 2.3.57 Autospeed reduces to 1mm/s with multiple objects

ciro1138, 2021-09-14:
The auto speed setting seems to be broken in comparison to 2.56.x versions. This set of parts in 2.3.56.1 runs at 2hours 36 minutes for both. in 2.3.57 the single larger part runs slower at 2 hours 37 and then jumps to 4 days with both parts as the speed gets reduced to 1mm/s. 

Let me know what you need additionally here are some screenshots. 
[image]
[image]
[image]
[image]

NameOfTheDragon, 2021-09-19:
#metoo - I was going to report this but hadn't gathered enough details yet. It seems to work sometimes but not others and I was trying to work out what the critical difference was. There is definitely something broken with auto-speed though.

lukeslaboratory, 2021-09-26:
Are your thin layers / gap fill settings also at speed 0? (autospeed) I also get this bug when I have thin walls or gap fill at 0, and also repeated by others in voron discord. Once i manually set a speed for gap fill and thin walls, it slices per normal.

supermerill, 2021-09-27:
@ciro1138  
I need your project file to reproduce.