supermerill/SuperSlicer #1564: error: ‘offset_ex’ was not declared in this scope; did you mean ‘offsetof’?

BosenSwe, 2021-09-14:
**Describe the bug**
When trying to build on Fedora 34 I get following error:
`/home/bosen/temp/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.cpp: In function ‘void Slic3r::GUI::add_brim(Slic3r::arrangement::ArrangePolygon&, const Slic3r::ModelConfigObject&, const Slic3r::GUI::Plater*)’:
/home/bosen/temp/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.cpp:83:34: error: ‘offset_ex’ was not declared in this scope; did you mean ‘offsetof’?
   83 |             ExPolygons brimmed = offset_ex(ap.poly, diff);
      |                                  ^~~~~~~~~
      |                                  offsetof
`

I think it's missing some #include, but I can't figure out which file that contains the offset_ex function.

BosenSwe, 2021-09-14:
I found out that it was missing #include "libslic3r/ClipperUtils.hpp"

supermerill, 2021-09-15:
I'll let it open a reminder to add the include