supermerill/SuperSlicer #1566: Layer Range Perimeters Breaks Solid Layers

randellhodges, 2021-09-15:
**Describe the bug**
Setting the perimeter count in a layer range override causes it to ignore the bottom solid layers

**To Reproduce**
Steps to reproduce the behavior:
1. Add a 50x50x10 box
2. Slice with 4 walls, 5 top/bottom layers, 40% infill, 0.2 layer height
3. View Gcode preview, looks good
4. Add a layer range override "Range 0.00-0.20", Add "Perimeters & Shell" which will add in your current top/bottom layers and perimeters.
5. Change the perimeters count to anything, increase by 1 or decrease by 1
6. Slice and view layer 2.

You'll notice that it only put in 1 solid bottom layer and started the infill at layer 2, but it should have 5 solid bottom layers.

**>> Project File <<**
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/7170670/Shape-Box.zip)

**Expected behavior**
Layers 2 thru 5 to continue to be solid

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.3.57.0

It also breaks if you go the opposite way of trying to do it. Slice normally with 1 single perimeter but add an override from 0.2 to 10 and set the perimeter count to 3. Starting at layer 2 the infill will not be solid.

supermerill, 2021-09-15:
When you add a layer range, it creates a new region.
The bottom layer count is only done inside a region, so the count break when the region change.
It's not simple to change that, as the entire way of creating shells needs to be rethought and rewritten.

randellhodges, 2021-09-15:
Interesting. Slicers are so intricate, I'd go crazy trying to keep it together.

What was trying to accomplish with this was just to emulate the single perimeter on top layer, but just for the bottom layer. Some models I like to have a high wall count, but on the bottom (which is the visible side) I'd like 1 (or maybe 2) walls.

In reality, what I'd like is feature that lets you set the wall count for top (expanding out your current 1 perimeter feature now) and bottom layers. Would that be any easier than to implement, or is it still part of the "entire way of creating shells needs to be rethought and rewritten"?

supermerill, 2021-09-26:
you want this feature request? #289

randellhodges, 2021-09-26:
That feature request would resolve what I was really trying to accomplish. A single perimeter bottom layer (like you have for the top) would be great. 

What would be even better is give the user the ability to specify how many perimeters on the bottom (and top). 1 is great, but sometimes 2 I think looks/works better.

closed 2021-11-07 by supermerill