supermerill/SuperSlicer #1567: Ability to enter the absolute overall print speed in mm/s

laptop-laptop, 2021-09-16:
**Is your feature request related to a problem? Please describe.**

Feature request: Ability to change the overall print speed to a specific speed in mm/s.

Asked Prusa support and three agents could not provide an answer on how to change the overall print 
speed of the i3MK3S+ to a specific speed in mm/s. For example, I want to change the overall print speed to
20%. Two person told me to use the dial to change it. However, the dial is in percentage, not in absolute
value. They also could not tell at what speed does such percentage is based on. They also mentioned about
changing the speed in the slicer but could not tell which parameters to change.

In Super Slicer, there are many speed settings. Some of them are also in percentage. If I want to change the overall print speed to a specific value, say at 20mm/s, which parameter(s) should I modify? As I don't know the default print speed of the printer, I cannot use percentage correctly. 

**Describe the solution you'd like**

In the slicer, allow the users to enter the desired overall print speed in mm/s.

**Describe how it would work**
See above

Area5142, 2021-09-17:
A suggestion:

If you like to change the overall speed of all print moves, the dial on the printer can be used or gcode M220 can be inserted into the print file. To reduce all speeds to 20%:

`M220 S20 ; Reduce print speed to 20%`

The effect of M220 is until changed or restart (power on/off).

I use this method to reduce travel speed for the first layer to prevent small details detaching from the build plate and stick to the nozzle. In SuperSlicer (or PrusaSlicer) this is the only way I can reduce travel speed for only first layer.

laptop-laptop, 2021-09-17:
I know about M220 and the dial but the problem is it is expressed as percentage. So percentage of which speed parameter? For example, if the printer prints at 100 mm/s, 20% means 20 mm/s. If the printer prints at 200 mm/s, 20% means 40 mm/s. So just M220 S20 could be speed other than 20mm/s.

If I want to print at a specific speed of 20mm/s, what should I do?

Area5142, 2021-09-17:
Absolute speeds (mm/s) can only be set in the g-code (by the slicer program).
There are more than one specific speed to set: Travel-speed, perimeter-speed, first layer speed, perimeter speed, infill-speed and a lot more.

Which speed do you want to set?

Slicer profiles can always be defined for different job types, setting the required speeds for that job type.
The physical printer may run slower than specified in the g-code due to physical limits as acceleration, motor power or limits in the firmware.

supermerill, 2021-09-17:
If you want the same speed for everything, set everything to 0, and set the max speed at the bottom (max_print_speed).
Set the max_volumetric_speed to 0 if you don't use it.

laptop-laptop, 2021-09-17:
Thank you. Is the special layer that is in between the top of Support and the Overhang surface (i.e. the special layer that is made up of the Overhang Perimeter and Bridge Infill) part of Support? If I want this special layer and support interface layers to print at 20mm/s, what should I do?

laptop-laptop, 2021-09-25:
Under High Range Modifier->Add Settings->Speed, there is a setting called "Default speed", what is that?

supermerill, 2021-09-27:
> Under High Range Modifier->Add Settings->Speed, there is an option called "Default speed". What is that for? Default speed for which setting(s)?

It's a typo, should be "Perimeter speed", will fix that.

> Case 1. [...] what should I put under Gap fill and Thin Walls?

I don't know, whatever your printer is capable of. Default values should be okay in most printers.

> In Case 2, [...] what should I do?

you can add all speeds in the modifier and set them to the value you want.

supermerill, 2021-09-28:
model.
support things are colored to support color

laptop-laptop, 2021-10-01:
Thank you. I can see that without support, the Overhang Perimeter and Bridge Infill parts are still there. So they became part of the model.

Here is the confusion: Supposing that I am printing an inverted L-shape consisting of a vertical and a horizontal bars. Now the Bridge Infill and Overhang Perimeter are added to the bottom of the horizontal bar. Does that mean SuperSlicer modified my model by: a) replacing the bottom flat surface of the horizontal bar by a series of Bridge Infill parallel lines along with an Overhang Perimeter as the outline of the rectangular shape or b) added an extra layer consisting of Bridge Infill parallel lines and Overhang Perimeter at the bottom of the horizontal bar? Accuracy is very important for me as I join various parts together.

supermerill, 2021-10-01:
It extrudes the unsupported alyer with more flow.
I'll add an option to get the bridge flow equivalent to the current layer height after prusaslicer merge.
Currently, you have to compute it yourself.
current 100% bridge flow is a circle of nozzle diameter diameter. If you want it to be of your layer height, you have to reduce the flow to get the bridge flow diameter equal to the layer height.

formula : ratio = 100 * layer_height² / nozzle_dimeter²
If you have 0.4mm nozzle and 0.2mm layer height, you have 100 * 0.04/ 0.16 = 25, so you have to set bridge flow to 25%

laptop-laptop, 2021-10-01:
Thank you. I am using a 0.4mm nozzle and 0.15 mm layer height. So ratio = 100 * 0.0225 / 0.16 = 14%?

If I use water soluble on support below the horizontal bar of the inverted L-shape object as well as on the layer with Overhang Perimeter and Bridge Infill to avoid seeing those parallel Bridge infill, the horizontal bar of the inverted L-shape model will be 0.15mm less thick?

supermerill, 2021-10-01:
yes
no bridge flow if using soluble support interface.

laptop-laptop, 2021-10-01:
Thank you. I will try both methods and see which is better for my case.

closed 2021-12-15 by supermerill