supermerill/SuperSlicer #1573: Infill Sparse Percentage box is not wide enough. Can't see the number

ChaosBlades, 2021-09-18:
**Describe the bug**
The location where you set the infill percentage is not wide enough to see or edit the number.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Print Settings
2. Got to Infill
3. Look at the drop down box right next to Sparse at the top of the page.

**Expected behavior**
The box to be wide enough to view and edit a value as high as 100

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Pop!_OS 21.04 x86_64 
 - Version: 2.3.56.9 (I have seen this bug in multiple previous versions)

**Additional context**
I am not certain if this is a linux specific issue but all other GUI elements seem to be correct.

supermerill, 2021-09-26:
I don't have this problem in my windows os.
You use kde, gnome?
what is the resolution of your screen? 1920*1080?
Do you have a "scaling" to show bigger text & icon? (something like 150%)

Can you take a bit wider picture like mine?
[image]

ChaosBlades, 2021-09-26:
I am using Gnome with a resolution of 3440x1440. My laptop also has the same issue at 1080p. Then my other laptop running Linux mint cinnamon also has the same issue. So it is not a resolution or desktop environment issue.
[image]

ChaosBlades, 2021-09-26:
Also, no I do not have any scaling set.

supermerill, 2021-09-26:
That's weird. The box are shorter than expected.
I'm going to increase only this field for now.
If you have some other too short field, tell me and I'll add some extra length for linux build.

ChaosBlades, 2021-09-26:
That is the only one with problems that I can find.

closed 2021-11-07 by supermerill