supermerill/SuperSlicer #1575: [FR] Start Z XY gcode change

qwewer0, 2021-09-19:
**Is your feature request related to a problem? Please describe.**
The slicer (both Prusa and SuperSlicer) first moves the nozzle down to the first layer height - right after the custom start gcode- then moves the nozzle in the XY. (If Z-lift is turned off)

**Describe the solution you'd like**
Add an initial Z-lift for the first Z XY move so the nozzle first moves up (away from the bed) then move to the needed XY position, then back to the first layer height. Exactly as with Z-lift enabled, but only for the first layer.

As gcode:

Current behavior:
```
(End of custom start gcode)
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-0.6 F3600
G1 Z0.2 F15000
G1 X54.1827 Y54.225
G1 X108.3654 Y108.45
G1 E0.6 F5400
;TYPE:Skirt
```

With the modification:
```
(End of custom start gcode)
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-0.6 F3600
G1 Z1.4 F15000
G1 X54.1827 Y54.225
G1 X108.3654 Y108.45
G1 Z0.2
G1 E0.6 F5400
;TYPE:Skirt
```

**Describe how it would work**
There could be an "Initial Z-lift" setting, where we would set the lift value, which would add a Z-lift at the first Z XY move, and 0 would disable it, which would make it behave like it does at the moment.

**Additional context**
https://github.com/prusa3d/PrusaSlicer/issues/4065
https://github.com/prusa3d/PrusaSlicer/issues/2931

supermerill, 2021-09-26:
retract_lift_first_layer ?

qwewer0, 2021-09-27:
> retract_lift_first_layer ?

Yes kind of, but it would be best if we could independently set the value for the first lift.

supermerill, 2021-09-27:
There is already an extra lift for the first layer (it's the height of the highest first layer).
I'll allow retract_lift_first_layer to be activated even with 0 lift / 0 retract.

This should be enough?

qwewer0, 2021-09-28:
> If this is enabled and the lift value is 0 or deactivated, then every first move before each object will be lifted by the first layer height.

Tested on [Nigthly windows x64 #452](https://github.com/supermerill/SuperSlicer/actions/runs/1279769267)
[Shape-Box.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7246184/Shape-Box.3mf.txt)
Why does it z lifts 0.6mm if retract_lift_first_layer[0] enabled and retract_lift[0] is set to 0? (layer height is 0.2)

And I think, the first lift need to be set independently from the normal z lift, so that it can be set to a higher than default value even if for the print there is no lift specified. So maybe a input box would be better instead of a check box and a input box.

supermerill, 2021-10-08:
> There should be a separate z lift setting just for the first move and a separate one just for the first layer, both with changeable values.

So this shouldn't be a "first layer z lift" but more of a "min z for travel", so it can be applied to all layers, to prevent a "clamp crash" in a second layer travel.

supermerill, 2021-11-02:
Should be up in the latest nightly.
Tell me if it's acting accordingly to expectations.

qwewer0, 2021-11-03:
> Should be up in the latest nightly. Tell me if it's acting accordingly to expectations.

Looks nice, but I found some things that is worth mentioning:

- If `lift_min` and `retract_lift` are both set, then the minimum lift in the gcode is the sum of those two
- If `lift_min` and `retract_lift` are both set, then on layer change the full z move is broken into two moves (`lift_min` and `retract_lift`)
- The `lift_min` is (sometimes) used inside of a single object, when moving from one feature to the other, e.g. perimeters to infill.
The tooltip looks logical, so I would think that `lift_min` wouldn't be used inside an object (but would be used from the brim/skirt if the `skirt_distance` is more than 0, which is the case right now)

>When an extruder travels to an object (from the start position or from an object to another), the nozzle height is guaranteed to be at least at this value.
>It's made to ensure the nozzle won't hit clips or things you have on your bed. But be careful to not put a clip in the 'convex shape' of an object.

- In the included 3mf file, there are layers when both z moves are used and some where only the `retract_lift`

In the video the layer change can be seen, which consist of two moves:
https://user-images.githubusercontent.com/57561110/140014199-a7275080-37ab-42ba-9266-46949cdcf1a3.mp4

[Shape-Box.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7465576/Shape-Box.3mf.txt)

I will do a more in-depth check later today, as I didn't had too much time in the morning.

@supermerill Thanks for all the hard work!

qwewer0, 2021-11-03:
This might be connected to the new changes in the code: https://github.com/supermerill/SuperSlicer/issues/1783

qwewer0, 2021-11-05:
If `lift_min` and `retract_lift` are both set, then the first z up move, right after the start gcode, will be broken into two moves.
e.g. if `retract_lift` is 1mm, and `lift_min` is 1.5, then first it will move up 1mm, then move an additional 0.5mm up, to an overall 1.5mm.

supermerill, 2021-11-05:
Can I have the project file?

qwewer0, 2021-11-05:
> Can I have the project file?

> [Shape-Box.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7465576/Shape-Box.3mf.txt)

supermerill, 2021-11-05:
outch, that commit broke evrything.

qwewer0, 2021-11-06:
> On the latest Nightly, if `lift_min` and `retract_lift` are both set, the first z up move, right after the start gcode, will be broken into two moves. e.g. if `retract_lift` is 1mm, and `lift_min` is 1.5, then first it will move up 1mm, then move an additional 0.5mm up, to an overall 1.5mm.

On the video, we can see the purge line (in green) from my start gcode, and the Z moves (in light blue).
In the project, the `retract_lift` is set to 1mm and `lift_min` to 1.5.

Additionally, if `lift_min` is turned off (0mm) or is equal or smaller than `retract_lift`, then the z moves will be 1.2mm (layer height is 0.2), and still be broken into 1mm+0.2mm moves.

https://user-images.githubusercontent.com/57561110/140597843-d2cb5889-2226-48d9-b772-06a39098d3b0.mp4

supermerill, 2021-11-06:
> be broken into 1mm+0.2mm moves.

Is it really a problem? the printer should make a continuous move.
Each z move is done in different (and far away) part of the code. I can't fuse them without creating problems.

qwewer0, 2021-11-06:
> > be broken into 1mm+0.2mm moves.
> 
> Is it really a problem? the printer should make a continuous move. Each z move is done in different (and far away) part of the code. I can't fuse them without creating problems.

Not really a problem, just reported it.
So I will close this then.

closed 2021-11-06 by qwewer0