supermerill/SuperSlicer #1576: GUI issues when in regular mode

LazaroFilm, 2021-09-19:
**Describe the bug**
The GUI doesn't show the right hand menu and sometimes the whole window when in regular GUI. The GUI works fine with the old Prusa Layout.

**To Reproduce**
From GUI in Old Prusa layout, set the GUI to regular, then switch around between tabs. 

**>> Project File <<**
Issue apparent with every projects. 

**Expected behavior**
GUI should display normally when switching tabs. 

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: MacOs Big Sur
 - Version 11.5.2

**Additional context**
Let me know what we need.

supermerill, 2021-09-26:
That's why I added '!! Can be unstable in some os distribution !!'
Maybe I should just remove the option on macos.
I can't fix it right now, I need a mac.

GarthSnyder, 2021-10-30:
@supermerill, do you have a theory as to what might be going on? I can take a look in the debugger, but I'm hampered by my rusty C++ and not being familiar with whatever cross-platform GUI is being used.

supermerill, 2021-10-30:
For this to work, I have to move the panel from the previous tab to the new one. it's a bit hacky, and each os has to be given the "please don't refresh' and 'please refresh now' at the good time.
tial&error is the only real method I have.

GarthSnyder, 2021-10-31:
OK, but where is the code that does this? I need something really basic like "The tab bar is a FooTabBar object. When you click a tab, it calls BlahBlahBlah. The actual panel rearrangement is done in SomeDumbFunction."

supermerill, 2021-10-31:
> OK, but where is the code that does this? I need something really basic like "The tab bar is a FooTabBar object. When you click a tab, it calls BlahBlahBlah. The actual panel rearrangement is done in SomeDumbFunction."

Most of the problem should lie in `MainFrame::init_tabpanel()` (src/slic3r/GUI/mainFrame.cpp:line670), you can easily spot macos-only debug lines. At line 701, it's the callback for when you click on a tab 'button' (or call a change via a function, like for when the slicing ends). It's inside the `if (this->m_layout == ESettingsLayout::Tabs)` that is the problematic code.