supermerill/SuperSlicer #1578: Prints created are smaller than originally designed

laptop-laptop, 2021-09-19:
**Describe the bug**

I noticed that the prints I have been making recently are smaller than my original design. In particular the surfaces parallel to the build plate shank by about 85-90% . Most probably an error on my side rather than a bug. Could you please confirm the following? I might have messed up something when I tried to resize the holes to fit other parts using the software versions 2.2.53 and 2.3.56.

Under Slicing->Modifying slices->XY compensation, I have Outer and Inner set to 0 mm by default. For First layer, I have -0.2 mm.

As for Vertical Hole shrinking compensation->XY compensation, I have -0.325 mm and Threshold with a default 100 mm^2.

As I understand, changing XY Compensation->First layer and Vertical Hole shrinking compensation->XY compensation only affects the hole size. The size of the entire model including the surface touching the build plate does not change. Am I correct?

supermerill, 2021-09-25:
Maybe your filament shrink when cooling, it's an intended behavior.
To compensate, you can use the "shrinkage" setting under filament (or scale your model).

laptop-laptop, 2021-09-25:
Thank you. Actually what do Vertical Hole shrinking compensation->XY compensation = -0.325mm and 
 Slicing->Modifying slices->XY compensation->First layer -0.2 mm respectively means?

supermerill, 2021-09-26:
`XY compensation=-0.2` will reduce the width of the object by 0.2mm on each side: if it's a (xyz) 20mm x 1mm x 10 mm, it will print a 19.6mm x 0.6mm x 10 mm object.

The first layer works only on the first layer.
The Vertical Hole shrinking work only for "holes" that are detected as convex.

When you look at a single layer, holes are empty areas that are surrounded by a not-empty area.

closed 2021-12-15 by supermerill