supermerill/SuperSlicer #1580: Missing geometry in sliced object

ManuelMcLure, 2021-09-21:
**Describe the bug**
When slicing a specific object, geometry is missing from the sliced results.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached project
2. Slice and look at the sliced preview - the volume under the buttresses will be empty.

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.
[LH_Green_Cape_body3.zip](https://github.com/supermerill/SuperSlicer/files/7199340/LH_Green_Cape_body3.zip)

**Expected behavior**
The object should be sliced as expected.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: Both 2.3.57.0 and 2.3.56.9.

**Additional context**
The original STL showed errors when loading into SuperSlicer - I used the repair function and the resulting .obj file (in the project I sent) no longer shows any errors.

ManuelMcLure, 2021-09-21:
Running the original Microsoft STL through the Microsoft 3D Tools web site repair tool solves the issue, so it seems like it's more of an undetected corruption that's causing the problem. Let me know if I should close this or if you want it left open so you can see if there's something that can be done to improve the algorithm.

supermerill, 2021-09-26:
I'm not touching the auto-fix stuff, i let prusa spend time on that.

closed 2021-09-26 by supermerill