supermerill/SuperSlicer #1583: Gcode seems to be at wrong x/y position

jesterhead82, 2021-09-22:
**Describe the bug**
When slicing for example the attached files the Gcode seems to be at wrong x/y positions, using 2.3.57
This happens also when i slice other objects.
FYI: This happens only with my second printer profile (first one is 1 300x300 printer, 2nd is a 180x180 printer), this does not happen on the first printer (here it gets centers in the gcode as in the plate view)

Plate:
[image]

Sliced:
[Mini Bear 2 Left Z-Axis_with Retract.zip](https://github.com/supermerill/SuperSlicer/files/7213589/Mini.Bear.2.Left.Z-Axis_with.Retract.zip)
[image]

GCode Preview:
[image]

Gcode In PS Gcode viewer:
[image]

[Mini Bear 2 Left Z-Axis_with Retract.zip](https://github.com/supermerill/SuperSlicer/files/7213595/Mini.Bear.2.Left.Z-Axis_with.Retract.zip)

jesterhead82, 2021-09-25:
Hello,

I found out that only the Gcode view seems to be shifted (for all prints), but when I print the sliced Gcode everything is fine.

supermerill, 2021-09-26:
maybe something with 32 bit precision somewhere... i'll check.

supermerill, 2021-09-26:
your start_filament_gcode
```
; Filament gcode
G0 Z0.2;Move to default first layer height
G92 Z0.2-0.04} ; Positive Value means lower z base
G0 Z0.2;Move to adjusted first layer height

; Filament gcode\nM900 K{if printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.6}0.12{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/}0.1{elsif printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{elsif nozzle_diameter[0]==0.6}18{else}0{endif} ; Filament gcode
```
look at the third line, there is an error.
do you mean `G92 Z{0.2-0.04}` ?

jesterhead82, 2021-09-27:
Typo on my side ;)

But still happening after fixing that.

supermerill, 2021-09-27:
> But still happening after fixing that.

are you sure? it fixes it for me.

jesterhead82, 2021-09-27:
Hmm, relselectedfilament type then it worked for me too, sorry about that

closed 2021-09-27 by jesterhead82