supermerill/SuperSlicer #1589: first layer support "brim"/interface uses inconsistent settings

nonchip, 2021-09-26:
**Describe the bug**

The first layer "brim" of the generated supports is coloured as a normal support, and uses most settings (such as spacing) from the normal support settings, but uses the interface pattern. this leads to useless unconnected outlines when the interface is set to concentric.

**To Reproduce**
Steps to reproduce the behavior:
1. Set a support spacing of >0
2. Set a support pattern of rectilinear
3. Set a support *interface* pattern of concentric
4. See error

**>> Project File <<**
[Shape-Box.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7231080/Shape-Box.zip)

**Expected behavior**
The pattern should be rectilinear, since it uses all the other settings for a normal (non-interface) support.

(also, maybe feature request worthy: additional settings for just this part might be helpful too)

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Version 2.3.57.0

neophyl, 2021-09-26:
PS support in the 2.3.x versions and earlier also put support above the model too.  Its one of the reasons that PS 2.4 (currently out as an alpha release) has had support generation revamped quite a lot.  Its under a new 'snug' support option.  Given time Super Slicer builds will become based on PS 2.4 and some of the support issues will get addressed that way.  

Also there no point trying to rewrite support in SuSi when the underlying code is going to be changed, so I wouldn't expect support to be looked at until after a merge with PS2.4.  Which wont be until PS2.4 is released as a proper stable version.   I'm sure SuperMerill will correct me if I'm wrong :)

supermerill, 2021-09-26:
I'll tag it 'prusaslicer' to say that i'll review it after the 2.4 merge