supermerill/SuperSlicer #1592: Is the tangential lines to hole issue caused by limitation of Slic3r?

laptop-laptop, 2021-09-26:
**Is your feature request related to a problem? Please describe.**

Over the years, there have been some posts on unwanted tangential lines to hole in the middle of flat surfaces. It seems to happen to different slicers such as PrusaSlicer and Cura. Is it due to limitation of Slic3r? Using SuperSlicer 2.3.56,  I have tried these two sets of settings:

1. Infill 100%, Solid Monotonic, Top and Bottom Monotonic (filled). Ironing post-process: All top surfaces. 

2. Infill 100%, Solid, Top and Bottom all Ironing. 

For each of the above two sets of settings, I have tried various values for "Flow rate" and "Spacing between ironing lines" but there are always two parallel tangential lines to each hole in the middle of the prints (please refer to attachment). Those lines form a band like a bandage with the hole in the middle. In person, it looks like the color/brightness within the band is different from the rest of the prints.

**Describe the solution you'd like**

Complete removal of those tangential lines and with ironing, smooth and solid looking surface with uniform color and brightness. 

**Additional context**
Please refer to attachment. Those two unwanted parallel tangential lines always show up. Region A seems to be of different color/brightness than the rest of the print (Region B).
[image]

supermerill, 2021-09-26:
I don't understand,
Do you have a project file showing the problem?

laptop-laptop, 2021-09-26:
There is no specific project file. Any hole on a horizontal surface has this problem. I just used default settings with the ironing settings mentioned in the first post. The photo may be difficult to see but if you compare it with my drawing, you can tell that the top and bottom tangential lines to the hole form a band of different color/brightness in Region A.

[image]

supermerill, 2021-09-26:
Can't reproduce
[image]
[1592.zip](https://github.com/supermerill/SuperSlicer/files/7232306/1592.zip)

laptop-laptop, 2021-09-26:
In the visualization, I also got very nice looking one like the one you posted. However, in the actual prints, they looked different. I think it might be due to the fact that when ironing, rather than moving the nozzle straight from one end to the other end of the square in Region B, the nozzle ironed one side of the hole first and then moved on to the other side in Region A.

supermerill, 2021-09-26:
that depends on your settings, can you post the project file that created the problematic gcode file?