supermerill/SuperSlicer #1593: Force the STL replace (force reload from disk)

robustini, 2021-09-27:
I often make some small changes to an STL, and having saved the project I would like to replace it while keeping the various additions I made, for example some modifier boxes.
But the annoying mistake appears that it is not possible to do it, because the STL is not exactly identical to the original one.
So I don't understand what this option can be for, since the STL is already in the project and it is not possible to do a reload from disk if it is minimally changed.
I therefore ask to be able to force this option, even if the STL has a different name.
The two STLs in the image are practically identical, in the new one I created a small hole in the lower part, and for that reason it does not make me replace the one inside the project, it's a big limit!

[image]

supermerill, 2021-09-27:
can't reproduce.
can you zip your porject and the stl to "reload" ?

The only thing that needs to be identical is the filename.

robustini, 2021-09-27:
Inside the zip there is the project and the STL on which I want to reload (and it doesn't). 

[ALA_SX_3.zip](https://github.com/supermerill/SuperSlicer/files/7237987/ALA_SX_3.zip)

randellhodges, 2021-09-27:
Might be related, but sometimes I might export something with a different name. When I reload, it gives me a prompt to choose the file, but if I choose the one with the different/proper name, it won't let me. I'm sure that's by design, but it would be nice to let me do it anyway.

robustini, 2021-09-27:
But it shouldn't do the name check either, it should always just do it!
If you are wrong you see it, there is no need for the software to warn you and not to do it.

robustini, 2021-09-28:
I believe that for @supermerill it is a matter of a moment to modify the code to disable this useless check.
It must load from disk regardless.

neophyl, 2021-09-28:
So much of this is going to have to be redone anyway when the merge against PS2.4 is done.  PS2.4 allows the user to reload a different file from disk anyway.

robustini, 2022-03-12:
I'm trying again with the nightly and it is disastrous.
If I drag the STL it puts it exactly where the other is, if I do a reload or a replace look what happens, I even go to move the modifiers.

https://user-images.githubusercontent.com/3864712/158010163-5559d020-48a5-42a1-a580-e3210f105685.mp4

supermerill, 2022-03-12:
can't reproduce

https://user-images.githubusercontent.com/6536403/158019437-593662d5-c28f-4d68-a5f0-5141781c013e.mp4

robustini, 2022-03-12:
> can't reproduce
> cant_reproduce.mp4

Try with this STL.

[FUSO05.ZIP](https://github.com/supermerill/SuperSlicer/files/8237722/FUSO05.ZIP)

supermerill, 2022-03-12:
can reproduce with this one

neophyl, 2022-04-12:
Well you can copy/paste the modifiers from one object to another already.  If you select them using the tree view on the right and cntrl+c you can select the recipient model and cntrl+v to copy them over.  The issue with them is that for some reason your objects when you look at the local coordinates have rotation values.  Objects with rotations don't react well to modifiers being placed afterwards.  Which could do with fixing admittedly but I get the impression its not going to be a simple thing.

Some of us use little 'libraries' of small objects marked as non printable and placed off the build plater which we use to store commonly used modifiers that can be quickly copied to what we are working on.  You just save it as a default_xx project and load it in and you get all your favourite settings and a bunch of pre-done modifiers you can reuse.

Unfortunately for anything that requires painting such as seams or supports as that is very object specific the same copy/paste cannot be used.  From what I understand you are designating faces from a list and so when you reload/replace a model it always resets that list.  Even when reload is the same identical model.

robustini, 2022-04-12:
Yes I have seen, but you have to select them one at a time and in any case it does not copy the height range modifier.
A nice copy and paste of everything in one go would be perfect.

neophyl, 2022-04-12:
You can copy and paste the height range modifier but that one has to be done on its own.  The others you can select multiple at the same time and copy/paste.

supermerill, 2022-04-13:
I've seen a difference between objects just loaded and objects loaded by opening a project: they don't have their transformation stored the same way, and so may not react the same way to a "reload".

the code is a bit complicated to track & understand.

robustini, 2022-04-13:
I think the problem is that.
The reload works if the body was originally rotated in the project using the slicer, it cannot work if the STL was rotated differently.

supermerill, 2022-04-13:
> this is not there in a saved project.

it's present, but not used anywhere (I had to use it for the seam modifier, and couldn't spot anything else that uses it).

For future reference: normal objects have their transformation matrix stored in `this->m_transformation`, objects loaded from a project have their original transformation stored in `this->source.transform`