supermerill/SuperSlicer #1596: Extruder decimals setting has no reset buttons

wlhlm, 2021-09-29:
**Describe the bug**
The configuration option "Extruder decimals" (Printer Settings > General > Firmware > Gcode precision) has no button to allow it to be reset to the default/previously saved value.

**>> Project File <<**
[ss-bug.ini.gz](https://github.com/supermerill/SuperSlicer/files/7249849/ss-bug.ini.gz) (exported config in case it's needed)

**Expected behavior**
I expect the configuration option to have the same reset buttons available as its sibling option "xyz decimals" (and pretty much any other slicer option).

**Screenshots**
[image]

**Desktop:**
 - OS: Arch Linux
 - Version: 2.3.57.1 8b942dd6b3c3d9dd7065eed313eea9d3dc65c6b7 (personal build of the Nigthly branch with gtk2)

supermerill, 2021-09-29:
Seems like i've created it with the idea of putting it into the "extruder" section but reverted back to general one... but forgot to change the type.

wlhlm, 2021-09-29:
It is actually still listed in the extruder section, but it's missing the reset buttons there as well.

supermerill, 2021-09-29:
Does it is useful to have precision per extruder? I don't think so... it's purely a firmware thing.
right?

wlhlm, 2021-09-30:
Probably. Klipper parses those values as Python floats, so limit for number of decimal places is pretty high.

Same data points on extruder precision: an Orbiter-like extruder (7.5:1 gear ratio) with 200 steps/mm, 1/16th microstepping, rotation distance of abour 35mm has a step distance of about 0.0015mm. There are worm-driven extruders out there with 20:1 gearing which comes out to a step distance of about 0.0003mm (20:1 gear ration, 200 steps/mm, 1/16th microstepping, 22.6 rotation distance).

I'm not knowledgeable enough about FDM to know if extruding to that precision make sense.

wlhlm, 2021-10-03:
Just built the latest changes from Nigthly branch and it's fixed now. Thanks!

closed 2021-10-03 by wlhlm