supermerill/SuperSlicer #1600: 2.3.57: First Layer XY Compensation Causing Shrunken Disconnected Infill

CCS86, 2021-09-30:
[image]

[1stLayerCompBug.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7262481/1stLayerCompBug.3mf.txt)

supermerill, 2021-10-01:
version?
can't reproduce on nightly.

CCS86, 2021-10-01:
2.3.57.0

CCS86, 2021-10-01:
I just imported that project file on a different computer, running 2.3.57.0 and it didn't reproduce either. 

I still have SS open on the original computer, exhibiting the issue. Is there anything I can gather for diagnostics?

supermerill, 2021-10-01:
it's not at the first layer but at 1.2 mm

supermerill, 2021-10-01:
that's because you set 0% in sparse infill ratio.
This deactivate all infill, but the one from bottom/top layers. That's why it's so thin, as it's the size of the first layer are + default anchor.

So set the `anchor solid infill by X mm` -> `defaut` to something very big like 10000

CCS86, 2021-10-03:
It still seems odd to me. 

If the number of layers for XY compensation is 1 or 2, there is no issue. But once it reaches 3, the infill becomes gapped. Shouldn't the infill area be calculated per layer, even for "bottom layers"? Why would a contracted layer 1 & 2 cause no change, but a contacted layer 3 cause this issue? Is there some hard coded layer 3 calculation, since 3 bottom layers is default?

[image]

supermerill, 2021-10-04:
> Shouldn't the infill area be calculated per layer, even for "bottom layers"?

it is

> Why would a contracted layer 1 & 2 cause no change, but a contacted layer 3 cause this issue?

because more layers are smaller, and when computing the infill of the next layer, the smaller bottom layer will be taken into account, hence the "too small infill".
The problem is the slope + no sparse infill. The XY comp is just exarcebating the issue.

> Is there some hard coded layer 3 calculation, since 3 bottom layers is default?

no

> anchor solid infill = 10000 does not change the result.

dang, I missed something.

CCS86, 2021-10-04:
With this model, by the 5th layer the walls are vertical. So, this is true even with 3 layers of XY contraction. The difference between perfect infill and disconnected infill, is solely on this XY compensation setting. How can a setting that only modifies the first 3 layers, break infill on the 8th layer?

Even if we were talking about layers within the scope of the XY compensation, my understanding is this: The model gets sliced first to establish the outer wall locations, then XY comp would be applied. then perimeters are generated moving inwards, when the correct number of perimeters are drawn you are (possibly) left with an infill area to calculate and generate. So, even if XY comp is applied, it seems like the infill should accommodate.

When using the new "height in layers" setting for XY comp, and pushing it above 2, it seems like the infill size gets stuck on the 3rd layer and applied to the remaining bottom layers.

https://user-images.githubusercontent.com/12439543/135859411-68a2beba-37e2-4dcc-a221-969318cbc9d3.mp4

supermerill, 2021-10-04:
> How can a setting that only modifies the first 3 layers, break infill on the 8th layer?

Even with 0 XY comp, the infill is disconnected. It's just less visible.
It's due to the 0% sparse infill "switch".

CCS86, 2021-10-04:
> > How can a setting that only modifies the first 3 layers, break infill on the 8th layer?
> 
> Even with 0 XY comp, the infill is disconnected. It's just less visible. It's due to the 0% sparse infill "switch".

Here is a result that suggests this is not the case. I compare switching sparse infill to 100%, vs leaving it 0% and disabling XY comp. Same result. Also, this is on layer 8, which is defined to be a bottom layer.

https://user-images.githubusercontent.com/12439543/135873457-9ea90416-acd7-4e35-8de6-0e2e5c200806.mp4

supermerill, 2021-10-05:
Took you .3mf, changed fill_density to 1%:
[image]