supermerill/SuperSlicer #1602: Large volume flow rate on steam

dmitrydprog, 2021-10-02:
**Describe the bug**
Large volume flow rate on steam. 

**To Reproduce**
1. Export my config bundle
[SuperSlicer_config_bundle.ini.zip](https://github.com/supermerill/SuperSlicer/files/7271691/SuperSlicer_config_bundle.ini.zip)
2. Load model (ring with two perimeters)
[Ring.stl.zip](https://github.com/supermerill/SuperSlicer/files/7271694/Ring.stl.zip)
3. Slice model (see on Volumetric flow rate)
[image]

**>> Project File <<**

[Ring.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7271696/Ring.3mf.zip)

**Expected behavior**
Volume flow rate in normal value.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version Version 2.3.57, also 2.3.56.8

**Additional context**

Video with print this area: https://youtu.be/87PR-eIniQw

supermerill, 2021-10-02:
you have a `G92 E0.0` at your custom gcode layer cahnge.
as you didn't set the printer to use relatice extrusions, the extruder has to make bigger and bigger extrusion to go back to the absolute value.

dmitrydprog, 2021-10-03:
Thanks you, it's worked.

closed 2021-10-03 by dmitrydprog