supermerill/SuperSlicer #1603: Overhang perimeter wrong layer height + other layer height oddities

SnowDrifterr, 2021-10-03:
**Describe the bug**
Overhang perimeters, and some internal bridging are an odd layer height

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'print settings'
2. Click on 'Perimeters&shell'
3. Scroll down to 'Overhangs'
4. Enable bridge speed and fan
5. Slice

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
Layer height in line with print settings

**Screenshots**
[image]

[image]
**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 22000.194

**Additional context**
[LayerHeight.zip.zip](https://github.com/supermerill/SuperSlicer/files/7272464/LayerHeight.zip.zip)

neophyl, 2021-10-03:
That’s normal for PS and Susi . For layers that are classed as bridging they can’t be squished onto the layer below so the software treats them as if they are the nozzle diameter in thickness. As that is how they will be extruded. The preview is showing that. It’s been that way forever and is one of the quirks of the software. 
It’s why the new Prusa Slicer alpha has a new setting called thick bridges. The ticked option is the old behaviour.  It’s also why I used to turn detect bridging off in PS as I usually found it printed better without. With Susi you have the bridging/overhang thresholds so you can tweak different values for the extrusion and for cooling so that it extrudes as a not bridge but is still cooled as one.

supermerill, 2021-10-03:
If you don't want bridge flow for overhang, set 0 to overhang->bridgeflow threshold.

tome9111991, 2021-10-03:
ok nice i didn't know that you would adjust the height with the flow

thanks.

supermerill, 2021-10-03:
there is nothing under it to flatten the extrusion.

closed 2021-12-15 by supermerill