supermerill/SuperSlicer #1604: Option to remove useless perimeter

tome9111991, 2021-10-03:
hello.
i have  a STL want print without support.
but i have some useless overhang perimeter they will print in air not connected to any perimeter.
[image]

maybe is possible to remove this?

project file
[OwlReDo1_fixed_sc.zip](https://github.com/supermerill/SuperSlicer/files/7273486/OwlReDo1_fixed_sc.zip)

supermerill, 2021-10-03:
I don't have a "remove unsupported island", should i add it?
[OwlReDo1_fixed_sc_noairprint.zip](https://github.com/supermerill/SuperSlicer/files/7273605/OwlReDo1_fixed_sc_noairprint.zip)

you can print your owl by rotating it a bit:

tome9111991, 2021-10-03:
rotating is nice idea.

but maybe i like "remove unsupported island" too.
maybe usefull for other project.
 if you have free time you can add it?

supermerill, 2021-10-03:
> free time

well, i'm taking days off to work on this project (thanks to supporters), and it's not enough.