supermerill/SuperSlicer #1607: Round corners observation, resolution bug?

hagan66, 2021-10-04:
I tried the new Round corners setting, and I think it is the first time I have ever seen the name plate on the Benchy sliced correctly in any slicer. Am I nuts? See attached sceenshots and prj file. :-)

[image]
[image]

[3DBenchy_round corners.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7277087/3DBenchy_round.corners.3mf.zip)

supermerill, 2021-10-04:
funny.
Does it prints well?

hagan66, 2021-10-04:
Actually prints great, do you have any ideas about this? :-)

[image]

supermerill, 2021-10-04:
The option change the way the polygon are shrinked. This have some positive effects, like this one. The most negative one is that it can increase a fair bit the computing time, because it creates much more points.

hagan66, 2022-07-16:
This phenomenon isn't working anymore :-) FYI