supermerill/SuperSlicer #1608: Is it possible to add an option to allow user to adjust the density of paint-on-supports?

laptop-laptop, 2021-10-04:
**Is your feature request related to a problem? Please describe.**

Even with paint-on-supports, often the supports are not dense enough to produce nice overhang surface.
For example, even I fully painted the surrounding of vertical holes and more support were added in addition to those when I chose Support Everywhere, the amount of support was not enough and vertical holes still deformed.

https://help.prusa3d.com/en/article/paint-on-supports_168584/

**Describe the solution you'd like**
Perhaps add an option to allow the user to adjust the density of the paint-on-supports 

**Describe how it would work**
Maybe add a slider and a box for user to enter the density

supermerill, 2021-10-04:
PS 2.4 will extensivly modiffy support & painting.
Can we re-review this issue after the merge?

laptop-laptop, 2021-10-04:
Sure. Thank you

laptop-laptop, 2021-12-14:
Hello, I tried PrusaSlicer 2.4.0-beta 4 but it does not seem to allow users to modify the density of the support via painting or other method. Is this feature hidden somewhere? By 'after the merge', do you mean after Prusa has released the official version of 2.4.0?

neophyl, 2021-12-14:
You have to set the support spacing to what you want.  Its global to the support settings.  
Painting or a modifier defines the area to be supported and then the other settings determine how that support is generated.  
Setting s like interface layers, interface pattern spacing, support spacing etc.  Its not hidden, unless you mean in plain sight.

laptop-laptop, 2021-12-14:
Thank you. Do you mean if I want to increase the density of the support, I increase the value in Support material-Options for support material and raft-Pattern spacing in 2.3.57.7? My main issues are: 1. Even I used the paint brush to paint the vertical holes and their surrounding, not enough support materials were generated and as a result, vertical holes still look deformed. 2. At the bottom of vertical walls, I have fillet. they don't look smooth curved edge because not enough support materials were generated. Under Preview, they look the same as if I did not manually paint the filleted edges. If these are the main concerns, is it better to use SS 2.3.57.7 or PS 2.4.0-beta 4? I don't know when they are going to release the official PS 2.4.0 and when merging with SuperSlicer will happen.

neophyl, 2021-12-14:
Just like on the Prusa forum , its best to attach a zipped up project file with an example of where you are struggling.  That way we ca adjust things and post back an example.

I do have a few comments though.  Fillets are NEVER going to look good with fdm printing.  No matter how much support you give them unless you have dissolvable supports and a zero contact distance.