supermerill/SuperSlicer #1609: Speed option for the first X layers above support interface.

AmyTheCute, 2021-10-04:
### Version
2.3.56

### Operating system type + version
Windows 10

Feature request:
Speed options for first few layers on top of support, additionally an option to turn off fan for those layers would be great

Reason: Soluble supports like PVA can work with abs provided that you print the first 2-3 layers over them VERY slowly (15mm/s) with no cooling fans, with this method I get good enough adhesion that long beams (32mm is the maximum i've tested) will not warp whatsoever, without these speed decreases I couldn't even print a 10mm long beam with abs.

I currently do this with heigh range modifiers, but with curved or sloped surfaces this means that I have to print a lot of my layers very slowly.

supermerill, 2021-10-05:
> A feature that lets me "Squish" the first layer a bit, as in print the interface 0.05-0.1mm into the ABS, so it sticks better!

why not just reducing the support top z-offset ?

AmyTheCute, 2021-10-05:
> > A feature that lets me "Squish" the first layer a bit, as in print the interface 0.05-0.1mm into the ABS, so it sticks better!
> 
> why not just reducing the support top z-offset ?

it won't let me put a negative value, that's a secondary small request tho! the speed alone would be very helpful

supermerill, 2021-10-06:
> it won't let me put a negative value

change the typ from "from filament" to "from plane"
A 0 offset from plane, menas that there is 0 empty space,exactly the same as for soluble filament, but without the switch.

AmyTheCute, 2021-10-06:
> > it won't let me put a negative value
> 
> change the type from "from filament" to "from plane" A 0 offset from plane, means that there is 0 empty space,exactly the same as for soluble filament, but without the switch.

that still won't let me add extra squish for adhesion.

But really the speed was my main request if that's possible!

supermerill, 2021-10-06:
> But really the speed was my main request if that's possible!

that's the problem with double-duty issues, i'm forgetting one of them, and the tracking isn't spot on.

> that still won't let me add extra squish for adhesion.

I can't let you do more, or the over-extrusion will destroy the print. If the squish you have from normal printing isn't enough, nothing will be.

AmyTheCute, 2021-10-06:
> > But really the speed was my main request if that's possible!
> 
> that's the problem with double-duty issues, i'm forgetting one of them, and the tracking isn't spot on.
> 
> > that still won't let me add extra squish for adhesion.
> 
> I can't let you do more, or the over-extrusion will destroy the print. If the squish you have from normal printing isn't enough, nothing will be. Looking again at it, it may be already over-extruding heavily, because of the bridge flow. Or send me your project, because i don't buy it.

my bad, i'll remove the squish request!

supermerill, 2021-10-06:
oh, i understand your problem.
uncheck the "do not support bridge" and it will stop making bridges over support.