supermerill/SuperSlicer #1610: Allow partial perimeter loop

burner-, 2021-10-05:
Currently if I use 0.6mm nozzle for 1,8mm wall it will do one perimeter to each side and then gap fill to fill that cap. It would be better to do 1 perimeter to another side and 2 to another. So basically currently when it calculate perimeter it will check that can it do full loop. If not it will remove whole perimeter loop. But better way would be that it do also allow partial perimeter loop where it goes as far it can and then stops when there is no space left to continue.  Many times this will remove need for gap fill.

neophyl, 2021-10-05:
The normal extrusion width for a 0.6mm nozzle is slightly wider, around 0.65 (check your extrusion width settings, we cant check as you haven't included a 3mf project file).  
Also walls have a slight overlap too  to make them stick together which is why its fitting in 2 perimeters but cant fit in a full third one so its doing gap fill in the space between.

supermerill, 2021-10-05:
> allow partial perimeter loop

so it's not a loop, but more like a gap fill?