supermerill/SuperSlicer #1613: BUG: Ball-shaped part not slicing correctly

Miksu80, 2021-10-06:
Top of the part is mis-sliced as per photo
File shows OK:
[image]

Sliced:
[image]

[image]

File:
[PenCase-Body.zip](https://github.com/supermerill/SuperSlicer/files/7293432/PenCase-Body.zip)

Config:
[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/7293496/SuperSlicer_config_bundle.zip)

Info:
Win10 / Superslicer Version 2.3.57

supermerill, 2021-10-06:
What's the program see when laoding the part (from meshmixer):
[image]
seems like some things are bad.
When the object is made solid by meshmixer, it can print:
[image]

I don't know how the object has been made, but there is some problems loading it.
Prusa is constantly working on it, and myabe itwill be better after 2.4 ...

Miksu80, 2021-10-06:
exported .stl -file makes this also :(