supermerill/SuperSlicer #1623: Crashes when Slicing

deathfrombelow, 2021-10-08:
### Version
2.3.56.8; 2.3.56.9; 2.3.57.1

### Operating system type + version
Windows 10

### Behavior
Program crashes when slicing.  on 2.3.56.8 and .9, it gets to about 90% complete,, preview window flashes blank, then crashes.  On 2.3.57.1, the progress bar completes then restarts, then repeats (like loading bar in office space), then crashes.  It doesn't matter how many objects are on the plate.  Same thing happens with 1 or 20 objects.

#### Project File (.3MF) where problem occurs
[v01_Main-parts.zip](https://github.com/supermerill/SuperSlicer/files/7314660/v01_Main-parts.zip)

Thanks

JohanAR, 2021-10-09:
Crashes on Ubuntu too, using 2.3.57.1
Looks like it slices the parts (progress bar going up several times) and then crashes about 50% into the "Generating g-code" stage

tome9111991, 2021-10-10:
is happend for me if i use.

Seam Position: Random
One-loop-Perimeter Seam Rear