supermerill/SuperSlicer #1624: Slicing error / random extrusion

JohanAR, 2021-10-09:
**Describe the bug**
Sliced g-code contains extrusions that don't correspond to the model geometry

**To Reproduce**
1. Load attached project file
2. Slice

**>> Project File <<**
[body.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7315279/body.3mf.zip)
(just rename to body.3mf it's not double-zipped)

**Expected behavior**
Slicing normally :)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 2.3.57.1

supermerill, 2021-10-09:
Found the problem.
It's a bug in clipper lib, when there is an horizontal line (as it scan by Y).
Finally found a way to mitigate it by using a little offset to prevent horizontal lines in the lib.
Big bug.

JohanAR, 2021-10-10:
Yep, the problem went away when I rotated the model a few degrees before slicing. Would've been much faster than manually editing the gcode file to remove the line, which I did :)