supermerill/SuperSlicer #1629: Auto-speed is broken again 2.3.57.1

NameOfTheDragon, 2021-10-10:
### Version
2.3.57.1

### Operating system type + version
Windows 10 Pro

### Behavior
Slice an object using a profile with auto-speed enabled.
[image]

Also note filament overrides:
[image]

Switch to GCODE view, and examine the print speeds.
[image]

The generated gcode has the entire model printing at 0.7 mm/s (which is way below the configured printer minimum print speed of 15 mm/s) - it should several orders of magnitude faster, especially the infill.
[image]

#### Project File (.3MF) where problem occurs
Use Calibration -> Calibration Cube -> Standard Cube, and set speeds as in the screen shots above.

neophyl, 2021-10-10:
The reason a project file is requested is that there are potentially other areas that can effect things. For example filament settings can over ride other speed settings. That’s why ALL the profile settings are needed.

supermerill, 2021-10-11:
You set 0 to gap_fill speed, bad idea as there are many very small extrusion (see tooltip).

jankolkmeier, 2022-01-14:
I noticed something (that I thought was odd) about autospeed for gap moves and had this image prepared for a new bug report, but I just found this discussion which seems to be talking about the same issue:

[image]

While this may be working as intended, I have to agree with others here that it is unintuitive.
My primary confusion is this: After setting gap speed to auto, I notice that print time has doubled. I want to inspect what's going on using the slicer preview. What I expect to see is that the percentage/time of "gap" extrusions has significantly increased (the white extrusions in the preview). Instead, I actually see that gap move time has decreased, but times of other moves, such as infill and perimenter, have significantly increased - although I made no change regarding those speeds.

The suggested change to the tooltip unfortunately does not clear up the confusion for me. In fact, it confuses me more: Why would thin gaps negatively affect the speed when using autospeed? Thin extrusions require less flow and thus should allow for faster moves? Apologies if I am missing or confusing some context required for understand this, @supermerill .

As a side note, the main reason I was interested in autospeed for gaps was to limit speed on extremely fat gaps, as I had previously exceeded my extruders capabilities with a fixed gap speed on those fat gaps.

supermerill, 2022-01-16:
Thin extrusion hit the max speed with very little flow rate.
The whole point of the auto-speed is to keep the same flow for the whole print if possible.
So it keep this extremely slow flow rate for everything.

jankolkmeier, 2022-01-17:
Thanks a lot for clearing it up, I didn't realize the purpose of autospeed is to prioritize *"constant pressure"* over *"as close to the max volumetric speed as possible"*. I was simply misusing this feature then. Reading it all back now makes sense, and I should have more closely read the tooltips under the  "Auto Speed (advanced)" section. 

I've tried to come up with a wording for a warning or popup that explains the issue without leaving room for confusion about the Auto Speed feature, but I'm afraid it is too wordy:

> **Auto Speed** will try to maintain a constant flow rate accross all print moves.
It is not recommended to include gap moves to the **Auto Speed** calculation (by setting this value to 0).
Very thin gap extrusions will often not max out the flow rate of your printer.
As a result, this will cause **Auto Speed** to lower the speeds of all other print moves to match the low flow rate of these thin gaps.

PS: To solve my original issue about fat gap moves specifically exceeding maximum flow, instead of Auto Speed I found that I can limiting maximum flow through filament_max_volumetric_speed.