supermerill/SuperSlicer #1698: Font size is unreadably small

jdblack, 2021-10-25:
**Describe the bug**
The font size for the superslicer UI is too small for me to read and I wasn't able to find a way to increase the font size.

**To Reproduce**
1. Open up superslicer 2.3.67
2. Wait until you get to 50 years old

**Expected behavior**
An option under Configuration-> Preferences -> GUI to make the font  as large as 200-300%

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: MacOS 11.4 Big Sur
 - Version [e.g. 2.3.67
 

**Additional context**
This is literally the first time I've used superslicer or, for that matter, a slicer other than Cura.   I usually have to set a font size of around 24 to  read UIs.

lucky62, 2021-11-22:
The same problem on Linux Mint.
[image]

neophyl, 2021-11-23:
I think this another one they have had with PS.  Might be worth checking what they did.
Sorry I currently dont have time to fins the issue numbers.  Busy at work :)

supermerill, 2021-12-01:
btw, did you used teh linux appimage of the gtk2-appimage? is the problem only on one of them?

lucky62, 2021-12-02:
> btw, @lucky62 did you used the linux appimage of the gtk2-appimage? is the problem only on one of them?

I didn't tried gtk2 version before..
Downloaded just now, and seems ok. 
I don't know which version I downloaded before, because I cannot find details..
File name was changed by AppImageLauncher and dialog "About" is showing the same info in both instances.
Both are the 2.3.57.6. No additional info.

Screenshot: Both instances running with open Preferences:

[image]

So, I will use the Gtk2 version and from my side the problem is solved. 
Thanks for help.

supermerill, 2022-03-14:
I added a preference to override the os scaling factor, but you will still have the os scaling for the menus and dialogs.

thisiscam, 2022-04-28:
Hi, I'm using the latest release on Mac and can't seem to find this option to scale font size?

supermerill, 2022-04-29:
oups, seems like I nuked it by mistake, i'll re-add it in next nightly.

timperlabs, 2022-05-05:
Just switching over to SuperSlicer from PS in hopes that the font size issue might be resolved here but I cannot find the configuration option under GUI.  I see only Tab Icon Size which doesn't seem to do much.   

I spent a few hours learning how to build the system though I know nothing about wxWidgets.  However, by brute force I bashed the font size mostly into submission by changing a few things including such as:

1) Change the default case to `return wxSystemSettings::GetFont(wxSYS_DEFAULT_GUI_FONT).Scaled(1.3);` in GUI_Utils.cpp.
2) Adding `.Scaled(1.3)` to definition of `m_normal_font`, `m_bold_font`, ... in GUI_App.cpp
3) Adding `Scaled(1.3)` to various calls to `wxSystemSettings::GetFont(wxSYS_DEFAULT_GUI_FONT)` throughout.  I am not sure why the code is littered with this anyway and not nicely factored out.

This seems to enlarge all fonts except for the ones on the left of the settings page next to the icons.  Those are very resistant.  This is important to me as I meet the OP's conditions and would like to continue using SuperSlicer when my RatRig arrives next month.   Please update on the status here and I'd be willing to help.  In my private build, you can see that most of the text is 30% larger now.

[image]

supermerill, 2022-05-06:
> Just switching over to SuperSlicer from PS in hopes that the font size issue might be resolved here but I cannot find the configuration option under GUI. I see only Tab Icon Size which doesn't seem to do much.

It's only in the latest nightlies (you can get them from the action menu):
[image]

timperlabs, 2022-05-06:
Thanks for the tip.  I tried the nightly dmg for last night and also for the one 3 days ago.  Both of them crashed on launch as follows.  

[image]

Is the fix in Master or some other branch?   I can build from there as I've built successfully using the BuildMacOS.sh script.

timperlabs, 2022-05-07:
I just tried the latest nightly Mac build again and it runs and shows the item in the Preferences box.  Changes to it are accepted and trigger a restart.  However, changing the value to 12, 15, 20 produces no change in font sizes and it still looks tiny.

timperlabs, 2022-05-08:
This might already be obvious or be known but I want to point out that the code for getting fonts in `GUI_Utils.cpp` is rather suspicious and looks like someone coded it only caring about Windows.  The method is supposed to care about dpi but does all sorts of calculations only if it's Windows.  Otherwise, it simply throws away the argument and returns `wxSystemSettings::GetFont(wxSYS_DEFAULT_GUI_FONT)` so it's not clear that the `DPIAware` template is going to help.

```
wxFont get_default_font_for_dpi(const wxWindow *window, int dpi)
{
#ifdef _WIN32
    // First try to load the font with the Windows 10 specific way.
    struct SystemParametersInfoForDpi_t { typedef BOOL (WINAPI *FN)(UINT uiAction, UINT uiParam, PVOID pvParam, UINT fWinIni, UINT dpi); };
    static auto SystemParametersInfoForDpi_fn = winapi_get_function<SystemParametersInfoForDpi_t>(L"User32.dll", "SystemParametersInfoForDpi");
    if (SystemParametersInfoForDpi_fn != nullptr) {
        NONCLIENTMETRICS nm;
        memset(&nm, 0, sizeof(NONCLIENTMETRICS));
        nm.cbSize = sizeof(NONCLIENTMETRICS);
        if (SystemParametersInfoForDpi_fn(SPI_GETNONCLIENTMETRICS, sizeof(NONCLIENTMETRICS), &nm, 0, dpi))
            return wxFont(wxNativeFontInfo(nm.lfMessageFont, window));
    }
    // Then try to guesstimate the font DPI scaling on Windows 8.
    // Let's hope that the font returned by the SystemParametersInfo(), which is used by wxWidgets internally, makes sense.
    int dpi_primary = get_dpi_for_window(nullptr);
    if (dpi_primary != dpi) {
        // Rescale the font.
        return wxSystemSettings::GetFont(wxSYS_DEFAULT_GUI_FONT).Scaled(float(dpi) / float(dpi_primary));
    }
#endif
    return wxSystemSettings::GetFont(wxSYS_DEFAULT_GUI_FONT);
}
```