supermerill/SuperSlicer #1737: Plater Grid "Split" at 45 Degrees

sojatay, 2021-10-30:
### Version
2.3.56.9

### Operating system type + version
Win 10 21H1 Build 19043.1319

### Behavior
After importing a valid config bundle from vanilla Slic3R, the build plate grid in the UI "split" at 45 degrees, and only displays a truncated plate. See attached picture.

SuperSlicer seems to be completely "stuck" like this now. I tried reinstalling, but that had no effect.

SuperSlicer Console is the same, but if I run superslicer_local-settings.bat, the plater grid appears normally.

Objects appear as normal on the plater, and can be arranged and sliced normally; the only real effect seems to be the GUI display of the plater grid

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[image]
N/A

supermerill, 2021-10-30:
can you add a cube, save the project file, zip it and drop it here?

sojatay, 2021-10-30:
Here it is. Fwiw, I just tried importing the same config bundle into PrusaSlicer, and it had the same effect. So it seems the problem is in the core PrusaSlicer, not in SuperSlicer
[Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/7445494/Calibration.cube.zip)

sojatay, 2021-10-30:
Update.....I deleted the Printer Settings that were imported, and that's cleared the problem. The "Print" and "Filament" settings are as imported I'll do some more fiddling around to see if I can find out exactly what caused the problem

supermerill, 2021-10-30:
Your bed_shape is `0x0,298x0,298x280,240x280`
maybe you want to change the last point by `0x280` ?

sojatay, 2021-10-30:
Yes. You are absolutely right. I have done that now and everything is OK. I don't know how the last parameter got there...
Thanks for your help. Sorry I wasted your time.

closed 2021-10-30 by supermerill