supermerill/SuperSlicer #1739: New feature : wave dense support

JackezBack, 2021-10-30:
**Is your feature request related to a problem? Please describe.**
Not easy to remove support from parts

**Describe the solution you'd like**
Playing with dense support layer, we could add z variation (parameter from 0.05 to 0.2) or pattern that make peak before part layer. This will reduce surface contact between support and part to be able to remove supports with ease. 

**Describe how it would work**
By applying a sinusoïdal fonction in z direction we can reduce surface contact between dense support and part. Each line could be glide by a half distance of peak to peak to get a more or less dense peak surface. 

**Describe alternatives you've considered**
-

**Additional context**
The feature could be drives with 2 parameters, the height of wave and the distance between 2 peak. 

I'm open to test and beeing in the loop to make it work. I'm a engineer working in the field of 3D printing for a while with 3D modelling skills, developing my own 3D printer.

supermerill, 2021-10-30:
a bit like the sawtooth pattern?
Speaking off, I have to improve it as currently it's flattened by the line next to it if too near.

JackezBack, 2021-10-30:
Yes like a sawtooth profil. That's it!

supermerill, 2021-10-30:
> Yes like a sawtooth profil. That's it!

Did you test it? What is wrong with it?

neophyl, 2021-10-30:
Merill I believe the OP means sawtooth in z axis not in x/y like the current version.  I suspect they think if theres intermittent contact at the interface layer the support will be easier to remove.

Personally I've found that increasing the interface layers to 4-5 make it thicker and therefore more rigid, actually helps support removal as the interface layer comes off in one piece.  At least when combined with a 0.2 contact distance. When its 1 layer thick you get lots of bits of interface stuck on to the surface so I don't expect a sine wave contact area to work well either.  

To the original poster, have you actually tried this method in practice by manually doing some gcode to test it or is it just a theory ?

supermerill, 2021-10-30:
> Merill I believe the OP means sawtooth in z axis not in x/y like the current version

The current sawtooth is doing skipes in z direction, or maybe I don't understand. But it need enough spacing between lines to not be flatten by the nozzle... It's far far from perfect, I have to dedicate some days to it to robustify it a bit.
[image]

[image]

neophyl, 2021-10-30:
My apologies.  I hadn't tried sawtooth, I thought it did similar to concentric/hilbert etc in the x/y plane with a zig zag pattern.  I think its the only one I haven't tried yet.

I think when my current print finsihes tomorrow I might have a play with that one.

JackezBack, 2021-11-01:
I tried sawthooth. It is not exaclty what I mean, but very close. Would it be possible to add smooth tooth a bit like this?
[image]
.

JackezBack, 2021-11-01:
I will try some test with FullControl GCODE

JackezBack, 2021-11-02:
Sorry I close by mistake...