supermerill/SuperSlicer #1742: First layer fan kickstart incorrectly interrupts start gcode

Ypsilantea, 2021-10-30:
### Version
2.3.57.3

### Operating system type + version
Windows 10 20H2

### Behavior
* _Describe the problem_
* First layer fan kickstart interrupts an intro purge line and causes a very large extrude
* _Steps needed to reproduce the problem_
* Have first layer fan on and a print move in your starting gcode, in my case G1 X60.0 E9.0  F1000.0 ; intro line
* _Expected Results_
* The final gcode outputs as expected
* _Actual Results_
* The slicer inputs an M106 command halfway into the G1 command (splitting it into a G1 X43.333 E6.50000) and then the ORIGINAL G1 command, causing the printer to extrude the full length of filament over half the distance. The final lines it generates:
* G1 X43.333 E6.50000  F1000.0 ; intro line
M106 S239.7
G1 X60.0 E9.0  F1000.0 ; intro line
*On 2.3.57.2, it behaves as expected, generating: 
*G1 X60.0 E9.0  F1000.0 ; intro line
M106 S239.7
*This happens even if "Only Custom Start G-code" is checked
*If you have multiple G1 moves in your start gcode, it only splits the last one
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
This isn't file-specific, so here's my full start gcode: 
G90 ; use absolute coordinates
M83 ; extruder relative mode

M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S170 ; preheat nozzle
G28; Home
G29; ABL
G1 X0 Y-3.0 F8000.0 ; go outside print area
M104 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; set extruder temp
M109 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; wait for extruder temp
G1 Z0 F360
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line

G21 ; set units to millimeters

G90 ; use absolute coordinates

M82 ; use absolute distances for extrusion

G92 E0.0

supermerill, 2021-10-30:
Can you post here your pàorject file? (save as -> zip -> drop it here)

supermerill, 2021-10-30:
new behaviour:
 * fan command from custom gcodes won't be moved.
 * when a fan is sent backward in time, and encounter a fan command from a custom_gcode, it will be print just after.
 * if "only custom start gcode", then the fan commands can't enter it.
 * 
Should be good?

Ypsilantea, 2021-10-30:
> 
> 
> Can you post here your project file? (save as -> zip -> drop it here)
> 
> On a more specific note, if your printer is with absolute extruder position, it doesn't prevent it to print the full line, it just adds a harmless fan command in the middle.
> 
> But I agree that it shouldn't touch it.

My start gcode has relative E on, which is probably what's causing it
Also, if I set my hotend spinup time to 0 it doesn't do it, which makes me think that from .2 to .3 either the "only for overhangs" box stopped working or start gcode started being treated as an overhang.

[Grip.zip](https://github.com/supermerill/SuperSlicer/files/7447455/Grip.zip)
.

supermerill, 2021-10-30:
> My start gcode has relative E on, which is probably what's causing it

The file you gave me hasn't it.  But whatever, not important.

> he "only for overhangs" box stopped working

doh! I inverted the check. It was doing "all but overhangs". Fixed

closed 2021-11-19 by supermerill