supermerill/SuperSlicer #1744: Milling feature: separate G-code

janrinze, 2021-10-30:
Pleasantly surprised to see the milling option!

I use a CNC router and wondered if it is possible to generate only the milling G-code.

As a next step i wonder if it would be possible in future to generate G-code for milling the 3D object from stock? It sort-of would be the inverse space of what is printed. Most desktop cnc routers have limited power so 'slicing' would definitely be the way to go.

I'm willing to test it on my router (will be using Styrofoam for testing)

supermerill, 2021-10-30:
Hi! happy to see someone else to test it.
Be careful, as it seems that some issues are here, and milling mistake are more dangerous!
#1426 
#1329
I didn't worked on these yet, but  now that I have a tester, I will.
I'll post here when work is done.

janrinze, 2021-10-31:
Currently the milling instructions in the G-code are per layer. This means that it can mill inside an object. However when milling is done separately this is not possible. Instead all milling should be done top down. I am looking into writing a small python script to extract milling parts from the G-code and reorder them top to bottom. If I can filter what is accessible from the bottom of an object i might split the G-code in Topview and Bottomview.
That way I can turn over the object and mill from the bottom.
more later..

supermerill, 2021-10-31:
What you want is a "milling" type of machine, like the "sla" mode.
Not for now, and I have the sls mode to do first (but they are very similar)

supermerill, 2022-09-09:
When new printers type can be added, and a mill printer is created, then I can think of adding this.