supermerill/SuperSlicer #1745: Brim Ears and Supports overlap as if the other weren't there.

Katherine1, 2021-10-30:
**Describe the bug**
When using both brim ears and supports, if the two end up on the same area of the model, they will both be placed down as if the other isn't there.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a model that has an overhang in an area that brim ears would cover
2. Add support to that area of the model
3. Set a brim width and enable brim ears
4. Observe the intersection of the two extrusions

**Expected behavior**
Either the first layer of supports gets replaced by the brim ear where the brim ear exists, the brim ear should wrap around the supports as a regular brim does, or the brim ear should not be placed in areas that already have supports. Regardless, the two extrusions shouldn't intersect like this.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 2.3.56.9

supermerill, 2021-10-30:
Can't reproduce
[image]

Can you upload your project file?

Katherine1, 2021-10-30:
I think I can do that. The part is a purchased model, but it is under the Creative Commons Attribution-NonCommercial-NoDerviatives licence, so sharing is technically allowed.

The part in this file comes from VectorFinesse's Head(amame) headphones.
[Spring Clamp.zip](https://github.com/supermerill/SuperSlicer/files/7447742/Spring.Clamp.zip)

supermerill, 2021-10-30:
you can remove now if it's need, and i can delete it when resolved.

Katherine1, 2021-10-30:
I'm not sure if putting it in a 3mf file constitutes modifying it as it was originally an stl, so I'll play it safe and remove it.

schattenphoenix, 2021-10-31:
My issue might be related.
When I have brim set and skirt loops not zero, both are created causing crossing lines.
It's not a huge problem as one or the other can be disabled but it feels like this might have some relation to your issue.

supermerill, 2021-10-31:
If you ahve this usse, check 'solid first layer' in the raft settings, and the brim should be good.

there seems also that 'skirt dist from brim' doesn't takes into account the support 'offset'

schattenphoenix, 2021-10-31:
[image]
This is what I currently have, but I assume it only takes effect if I add raft layers (at least 1)?

With these settings the "issue" is not fixed.
It's not a real issue, I was mostly wondering if this could not be solved more elegantly 👍

supermerill, 2021-10-31:
> With these settings the "issue" is not fixed.

then it's not the same issue.
I don't know what you want to say by "causing crossing lines".

closed 2021-11-19 by supermerill