supermerill/SuperSlicer #1748: Using a 0.3 height profile with a 0.4mm nozzle raise the error "layer height can't be greater than max layer height"

jfestrada, 2021-10-31:
**Describe the bug**
Using the Vertex profile (any machine) for 0.4mm nozzle and layer height 0.30mm is raising the error "Layer height can't be greater than max layer height".
The issue only happen with the version 2.3.57.3 using the 2.3.56.x and the same profile is working fine 
I have tested on windows and linux with the same results.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a the Vertex XL machine with nozzle 0.4
2. Add an stl
3. Choose the profile "0.30mm DRAFT"

**Expected behavior**
Slice the stl as usual

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux and Windows 10
 - Version: 2.3.57.3

tome9111991, 2021-10-31:
hello
you have tried to edit the "Layer height limits" ?

Printer Settings -> Extruder

jfestrada, 2021-10-31:
> hello you have tried to edit the "Layer height limits" ?
> 
> Printer Settings -> Extruder

You were right, that's the solution, I suppose that this check was not done in previous releases.

closed 2021-10-31 by jfestrada

supermerill, 2021-10-31:
yes, it's a new check, to avoid more problems.