supermerill/SuperSlicer #1750: Option to Tie more setting together and recalculate

inspectionsbybob, 2021-10-31:
This is a FEATURE request

There are many settings that derived from each other and have understood relationships as a percentage.  Speed and speed of Infill, support, bridge, etc, and many more on the SPEED page.  If the PERIMETER speed is the "Default" have an option to recalculate all of the other settings in the section to be a PERCENTAGE  of the Perimeter speed.

**Describe how it would work**
Put a button in the upper right corner of the "Speed for Print Moves" box labeled "CALCULATE."  Clicking that box would re-calculate and populate all the speeds on the page to the default percentages of the Perimeter Speed. 

**Describe alternatives you've considered**
Calculating and Editing each and every value to be consistent with the perimeter speed

**Additional context**
When Changing Nozzles, the Speed is dependent on what flow the printer can force out before it starts slipping.  Going from a .40mm nozzle to a .25mm nozzle requires about 1/3 the speed on everything... and that's a lot of edits...

would LIKE if the Speed was originally 60mm/sec and I edited it to be 20mm/sec, everything would be updated to represent the appropriate speed for that function based on Perimeter (default) speed.

neophyl, 2021-10-31:
This is already possible in the slicer.  There are 2 methods -
1- You could just set all your speeds to zero, which then already does calculate the speed then based on the default but using inbuilt percentages which may not suit some

2- or more usefully set them to % values and then all you have to change is the default speed and the others will be set.  

Remember the profiles are taken from external sources, if they don't suit you then its usual to tweak them until they do and then save your own custom profiles.  Just save yours with percentages.

supermerill, 2021-10-31:
You may want a "default speed" in the printer tab and allow the first speed column to be able to be a % of it?

inspectionsbybob, 2021-10-31:
Yes, and many other speed related options as well.

If EVERYTHING could be a % then you wouldn't need a calculate button