supermerill/SuperSlicer #1752: Appdata directory

tome9111991, 2021-10-31:
Hello.
im not sure it allready possible or its a Feature request.

how about if you can select the app data, i.e. all settings including config, the directory where it is saved?

so I could store all the settings from my NAS and synchronize several PCs.

supermerill, 2021-10-31:
You can access the dir via "help -> show conf folder"
If you have a "configuration" dir next to the exe, it will be used instead
You can choose the config dir via the `--datadir` command option, look at the superslicer_local-settings.bat in the win release

closed 2021-10-31 by tome9111991

tome9111991, 2021-10-31:
Thanks works so far.

Maybe some day possible to change path via GUI?

supermerill, 2021-10-31:
yes. There is a bit update for setting sync planned.

stevos758, 2021-11-02:
I was directed here from one of my questions asking the same thing. I am not farmiliar with how to set this up. Is there any documentation... Explain it to me like im 2. LOL

PJawZK, 2022-02-26:
In case someone else is looking for this, and don't understand what [supermerill](https://github.com/supermerill) means.

Here is a pic on how to set it up, as I figured he meant by using --datadir next to exe.

- You first create a shortcut from the superslicer.exe file.
- The you open the shortcut up, and locate the "Target:" bar with the folder structure text in it to locate the .exe.
- At the end of the current text inhabited in "Target:", add the "--datadir" followed by your folder structure/"link"

as an example;
C:\SuperSlicer_2.3.57.11_win64_220213\superslicer.exe --datadir C:\SuperSlicerConfig

-Where "C:\SuperSlicerConfig" is the location of your own slicer config files

I hope this clarifies things for other, more novice users, out there.

Happy 3D printing!
// --JawZ--

[image]

supermerill, 2022-03-01:
>  he meant by using --datadir next to exe.

To clarify, you have two options:
1) create a directory named "configuration":
[image]

2) use the --datadir option (takes precedence over everything). You can use the shortcut creation method by @PJawZK 
I advise to put the config path inside double quotes to avoid problems
`C:\SuperSlicer_2.3.57.11_win64_220213\superslicer.exe --datadir "C:\users\john smith\SuperSlicerConfig"`

axg20202, 2024-02-12:
I came here looking for this info. Just to add, if your alternative config folder is a network folder, enter as follows in shortcut Target:
"C:\Program Files\SuperSlicer_2.4.58.5_win64_220924\superslicer.exe" --datadir "\\NAS\SS_shared_configuration"