supermerill/SuperSlicer #1753: QUESTION about feasibility or Feature

inspectionsbybob, 2021-10-31:
This is more of a is it possible or is someone already doing this...

I want to INCREASE EDGE SHARPNESS.

For example, I have a Model of a Flower Bud vase.  Looks OK, but the individual Petals blend together and the transition is "Smooth".  I would like to AUTOMATICALLY find edges, and make them sharper/deeper/non-filleted.

[image]

Might be a script that uses Meshtools... not sure

supermerill, 2021-10-31:
It's more a modeling prpblem, no?

inspectionsbybob, 2021-10-31:
Many of the Thingaverse model are "smoothed" before we get them... this
would make existing "smoothed" models really pop

supermerill, 2021-10-31:
It's out of scope for me, sorry.