supermerill/SuperSlicer #1754: G-Code preview, Skirt and Support Material feature types swapped

keithstric, 2021-10-31:
**Describe the bug**
A clear and concise description of what the bug is.

When using the g-code preview, toggling Skirt and Support Material feature types, it seems these 2 options are swapped

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a model and include a skirt
2. Go to g-code preview
3. Uncheck all Feature types
4. Check Skirt --> The skirt doesn't display, however Support Material does
5. Uncheck Skirt
6. Check Support material --> The skirt is displayed

**>> Project File <<**
[xyzCalibration_cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7449351/xyzCalibration_cube.3mf.zip)

**Expected behavior**
Selecting Skirt in the g-code preview feature types should show the skirt
Selecting Support in the g-code preview feature types should show the support

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 11.5.1

**Additional context**
Add any other context about the problem here.

supermerill, 2021-10-31:
Ironing was missing. I should have missed it when merging with PS 2.3.

supermerill, 2021-10-31:
ps: isn't it overkill to use both ironing methods?

keithstric, 2021-10-31:
@supermerill probably, I've been using this for all of a few hours trying to match my Cura results with SuperSlicer. Not to mention I'm still kind-of a noob when it comes to 3d printing, but certainly trying to learn. 

So, since my noob status has been stated.... which 2 ironing methods are you speaking of?

As an aside, great turn-around time on fixing the bug. Not many products will fix a bug hours within receiving it. Thank You!

supermerill, 2021-10-31:
There is an "ironing" pattern for top infill. It's the legacy option, that doesn't add more plastic.
There is the "ironing post-process" that adds a thin layer of ironing over some top infill & perimeter. This is a new method from PrusaSlicer. When using it, you're better served by using monotonic for top pattern, as it's quicker.

closed 2021-11-19 by supermerill